exelban/stats #559: can't shown sensor reading and some setting fields

hungrysnake, 2021-07-26:
**Describe the bug**
A clear and concise description of what the bug is.

1. the sensor can't be shown
[image]

2. the setting of some items like label, background can't be shown (there should be some text field of button at the right side of the items)
[image]

**Details:**
 - Device: [e.g. Macbook Pro 2016] iMac mid 2010
 - MacOS [e.g. 10.15.5] High Sierra 10.13
 - Application version [e.g. 2.1.11] 2.6.2
 -

exelban, 2021-07-26:
Hi. Thanks for reporting.
I will try to fix it.

hungrysnake, 2021-07-26:
more info about the display issue. it seems all the sensors are working but not properly shown in the setting page.
the gif below shows the sensors reading the the setting page actually responding to click for enabling/disabling the menu bar items.

[image]

exelban, 2021-07-26:
The problem is with a toggle button. It does not visible in macOS 10.13. Will try to fix it today.

exelban, 2021-07-26:
Could you please check if this build will have missing buttons in the settings?
[Stats.dmg.zip](https://github.com/exelban/stats/files/6879586/Stats.dmg.zip)

hungrysnake, 2021-07-27:
thanks for making a fix.
the toggle checkbox can be shown now must slightly misaligned 

[image]

closed 2021-08-01 by exelban