exelban/stats #561: not able to set fan speed

hungrysnake, 2021-07-26:
even the hdd fan speed is set to 1100 but it is still running at full speed 6300rpm

[image]

exelban, 2021-07-26:
What if you change the speed manually?

exelban, 2021-07-26:
Sorry, but macOS 10.13 is not maintainable. Yes, it's supported, but will not receive updates or fixes.
It could be a lot of modules which does not work on 10.13. But it will be not fixed (only if someone will propose patches throw PR).

closed 2021-07-26 by exelban

hungrysnake, 2021-07-26:
> What if you change the speed manually?

yes, I manually set it to 1100 but no effect.

hungrysnake, 2021-07-26:
> Sorry, but macOS 10.13 is not maintainable. Yes, it's supported, but will not receive updates or fixes.
> It could be a lot of modules which does not work on 10.13. But it will be not fixed (only if someone will propose patches throw PR).

thanks for looking into it. sadly my iMac mid 2010 is only supported up to osx 10.13

exelban, 2021-07-26:
I suppose macOS 10.13 has different SMC keys for fans control. Or these keys have a different type. Give me know if you will figure out how it could be fixed.

hungrysnake, 2021-07-26:
I've just tried google a bit and use this freeware to set Mac FAN on my mid 2010.

[image]