exelban/stats #565: Feature Request: Custom colors

ivnnv, 2021-07-31:
Hi,
First of all, thank you for creating this app and sharing it with the community, I think this is finally the one final app that overcomes the old atMonitor once for all!

This is just a small feature request, could the app add color customization for meters? Just as Menumeters does, so we could change for example the "blue" to "green" (thats actually the only thing Id change hehe):

Stats:
[image]

MenuMeters with custom colors:
[image]

exelban, 2021-07-31:
Hi. Thanks for the proposal.
I think it will be a nice feature.

rswail, 2021-09-17:
add a vote to this issue, but primarily to make the CPU/memory/disk graphs have a "dark" mode coloring scheme/background.

Great app though, perfect for replacing previous tools :)

macbugs, 2022-03-19:
> **_primarily to make the graphs have a "dark" mode_**

I definitely agree this is needed, for example, compare MenuMeters 8-core CPU monitor on left, Stats GPU indicator on right[image]
...but also notice how the 'Stats' graph does not utilize the full amount of vertical space in menu bar (which is 22 pixels)

macbugs, 2022-03-19:
> **_primarily to make the graphs have a "dark" mode_**

I definitely agree this is needed: compare MenuMeters 8-core CPU monitor on left, Stats GPU indicator on right...
[image]
...but also notice how the 'Stats' graph does not utilize the full amount of vertical space in menu bar (which is 22 pixels)

closed 2022-10-30 by exelban