exelban/stats #570: Switching fan back to manual mode doesn't apply the previous RPM

Atemu, 2021-08-02:
**Describe the bug**
A clear and concise description of what the bug is.

When swtiching off of manual mode (i.e. to cool down the system after an intensive workload) and back, the manual setting doesn't get re-applied. You have to slide the slider or click one of the limits for the setting to be applied.

**Details:**
 - Device: MBP2018
 - MacOS: 11.5.1
 - Application version: 2.6.3

exelban, 2021-08-02:
Hi.
Do you switch from manual mode to automatic?

Atemu, 2021-08-02:
Ah, sorry; I switched from manual to full-power and back.

closed 2021-08-02 by exelban

Atemu, 2021-08-02:
I did not mean that the issue has been solved, I was just describing my situation in more detail. Sorry for the confusion; what I wrote is ambiguous.

I just tested switching to and from auto and that's also broken, so the original bug description is still correct.

exelban, 2021-08-02:
When you switching to full power it means that the app automatically sets the max speed of the fan. It does not mean that when you switch to the manual mode, it will change value. Because nothing was changed.

When you switching to automatic mode, it means that fans are controlled by the system. And system decides what fan speed needs to be set.

exelban, 2021-08-02:
What exactly is broken?
When you switching back to the automatic mode, the system does not control the fans?

Atemu, 2021-08-02:
1. Be in manual mode and set a specific speed
2. Switch to auto or full mode
3. Switch back to manual mode

Expected behaviour: The previously set manual frequencies are restored.

Actual behaviour: The behaviour of the previously selected mode remains active until manual frequencies are dialled in again.

exelban, 2021-08-02:
Nope. Switching between modes does not mean that the fan speed will be touched (only in the automatic, because in this case system control the fan).

When you switching from auto -> to manual you need to set the fan speed.

Atemu, 2021-08-03:
So this is intended behaviour?

exelban, 2021-08-03:
Yes. Because you just switching the fan mode.