exelban/stats #571: High cpu use in last version

ialiendeg, 2021-08-02:
When trying to click an applet to open the menu, the app goes high on cpu and activity monitor shows app is not responding.

I'm on MacOS High Sierra (v 10.13.6) in a Macbook Pro 2011.

exelban, 2021-08-02:
Hi. Thanks for your response.
Do you try to click on the menu bar item? Does it happens when you click on some specific item, or any of items?

There were no changes (between v2.6.2 and v2.6.3) that could provide this bug.

ialiendeg, 2021-08-02:
As far as I see, network, disk usage and sensors applets are ok. Cpu and Ram applets trigger the high cpu problem when clicking on them.

exelban, 2021-08-02:
Does only the Stats app start using a lot of CPU?
Or maybe `/bin/ps` or `ps` too?

ialiendeg, 2021-08-02:
Nope, the app starts ok, and ps is working good too. I even tested activity monitor and htop and no problem.

exelban, 2021-08-02:
Could you run `defaults delete eu.exelban.Stats` in the terminal and try again?

ialiendeg, 2021-08-02:
The problem is still happening after deleting the app defaults.

ialiendeg, 2021-08-02:
It's not critical for me, I'm using this old laptop while my new one is at service, so I can use the old version meanwhile.

But tell me if I can test something more for you.

Btw, congratulations for the app, very nice and useful :)

exelban, 2021-08-02:
I cannot imagine what you can check. I take a look at the changes from the previous version. And there were no changes which can effect in this way. The only you can do it's to run the Xcode and check where is the problem)

10.13 is not supported by me. So you can close this issue if it's not critical for you. I don't have a mac with 10.13 to debug it.

ialiendeg, 2021-08-02:
Closing then, thx for the help anyway :)

closed 2021-08-02 by ialiendeg

exelban, 2021-08-02:
If I will find macOS 10.13 where I can run XCode I will take a look at this.

ialiendeg, 2021-08-02:
Ok, I'm a developer, but with Ruby on Rails, not OS X apps.