exelban/stats #572: Add label option for network graph in menu bar

simondemartini, 2021-08-03:
The network chart doesn’t seem to have the same option to have a label included in the chart that the CPU, RAM and other line charts to. Can a label like “NET” be added to match?

[image]
For example: The network chart looks out of place here without a label.

There is an existing option for a separate “NET” label widget though, but that is separate from the graph widget so it has more spacing and doesn’t match the others.

wfrank21, 2021-08-03:
You can add the NET label in the Network modules settings. 
[image]

exelban, 2021-08-03:
Hi. Thanks for the report.
Yes. The network chart does not have this option. Because CPU/RAM it's a different widget than in the Network. But I think it could be added to be more uniform.

@wfrank21 yes, but @simondemartini talking about label near chart. It's has a little bit different view than the `label` widget. Because on macOS 11+ the widgets have internal paddings adding by the system. So label widget and label near widget look different. There is no this problem on < macos 11, because there are no paddings between widgets in the menu bar.

Label widget:
[image]

Label option in the widget:
[image]

closed 2021-08-15 by exelban