exelban/stats #573: Feature request: Remember manual fan setting

Atemu, 2021-08-03:
Currently, when you switch off manual mode and back, the previous manual settings are lost and not re-applied. https://github.com/exelban/stats/issues/570

It'd be very useful if stats could remember the manually set frequencies and re-apply them whenever manual mode is activated.

A related issue is that manually set fan speeds seem to be lost when the machine sleeps; that should be fixed too for this feature to make sense.

exelban, 2021-08-03:
You could ask in the previous ticket if it will be changed. Because the answer is no. There will be no memory for fan speed. To prevent a situation when you yesterday set the fan speed to max. And today when you switching back from auto to manual mode it spins your fan to max.

exelban, 2021-08-03:
Maybe it could be realized as an option in the settings. Which is disabled by default.

closed 2021-08-22 by exelban