exelban/stats #575: CPU frequency doesn't show

Accoustika, 2021-08-05:
Even though I enabled the CPU frequency in the options for CPU, I still can't see it when I click on CPU in the menu bar. Temperature and Percentage show normally but Frequency doesn't show any value, just an empty chart

Accoustika, 2021-08-05:
[image]

exelban, 2021-08-06:
Hi, Thanks for reporting.
Do you have installed IPG (intel power gadget)?
If yes, could you check the version of IPG?

Accoustika, 2021-08-06:
Hi,
I didn’t know I needed to have Intel Power Gadget, I was wondering what IPG stood for in the options, haha!
I’ve had it before but when I upgraded to Big Sur it stopped loading so I stopped using it. I just downloaded and installed the latest version, 3.7.0 and now I can see the core frequency in Stats! Thanks!

exelban, 2021-08-07:
Ok, so the problem is in the settings. When you have IPG installed, you have the option to enable/disable CPU frequency. But when you uninstalled it, the option is not showing anymore. I will take a look if I could do something, but the fastest fix is to reset the app settings: `defaults delete eu.exelban.Stats`.

closed 2021-08-07 by exelban