exelban/stats #583: The settings window is not compatible with macOS dark mode.

hazarek, 2021-08-12:
**Describe the bug**
The settings window is not compatible with macOS dark mode.

[image]

**Details:**
 - Device: MacBook Pro 15" (Late 2016)
 - macOS Big Sur (11.5.1)
 - Application version 2.6.3

exelban, 2021-08-14:
Hi. It's the design of the app. It does not depend on the color mode.

closed 2021-08-14 by exelban

hazarek, 2021-08-16:
> Hi. It's the design of the app. It does not depend on the color mode.

But why? Forcing everyone to use the light mode is not design.

hazarek, 2021-08-16:
https://www.wired.com/story/its-not-a-bug-its-a-feature/

exelban, 2021-08-16:
You could propose a new design.

hazarek, 2021-08-16:
xcode supports both modes automatically. What do you mean by design?
# Supporting Dark Mode in Your Interface - Article
https://developer.apple.com/documentation/uikit/appearance_customization/supporting_dark_mode_in_your_interface

Angelk90, 2021-09-20:
@exelban: So you have chosen to be in both light and dark mode, does not look to change?

exelban, 2021-09-20:
@Angelk90 I'm ready to add support for light/dark modes to the settings. But I need a UI that will look nice in both modes. The existing one does not conform to this requirement(

Angelk90, 2021-09-20:
@exelban : Based on the App Store, I think we should get something like style.
https://user-images.githubusercontent.com/20476002/134067591-2c2367dc-5f29-4c37-b5c3-1c8f33d22b90.mov

exelban, 2021-09-20:
@Angelk90 it's not only about changing the colors from light to dark. You cannot take anything that was designed for one color styling, change the colors to the opposite, and expect that everything will look fine.

I'm not good at UI (styling), but for me changing the colors to dark in the existing design makes settings look terrible. Maybe someday some designer will come and help with app design)

Idomo, 2021-12-18:
@exelban I’d like to help you with the Dark UI, just let me know how.

rakleed, 2022-01-23:
@exelban can you leave this issue open?

hazarek, 2022-03-09:
@exelban Why is this issue closed? Maybe the issue needs to be visible for a volunteer person to do it. This is a public repo not private.

exelban, 2022-03-09:
@hazarek hi, you are just spamming this thread?
As I say, the app has some design (even if I do not really like it). And it is not compatible with dark colors. For this feature, the app must have a little bit different design, such as a menu, buttons, list of widgets, etc.
If someone is ready to do this (a new design), you are welcome. But please do not spam just because you want.