exelban/stats #584: Feature Request: Locking individual fan speeds.

hazarek, 2021-08-12:
**Describe the bug**
Adjustment feature with a single bar or single menu button, not separately.
[image]

**Details:**
- Device: MacBook Pro 15" (Late 2016)
- macOS Big Sur (11.5.1)
- Application version 2.6.3

exelban, 2021-08-14:
Hi. Thanks for the feedback.
Could you please explain what do you mean?

hazarek, 2021-08-16:
> Hi. Thanks for the feedback.
> Could you please explain what do you mean?

Changing the right and left fan speeds using a single bar. At once.

exelban, 2021-08-16:
I do not see a reason why it's needed. And don't know how it's could be implemented in UX.

hazarek, 2021-08-16:
nevermind, You make me feel so tired

closed 2021-08-16 by exelban