exelban/stats #589: Feature Request: Display multiple BT Devices

wfrank21, 2021-08-15:
It would be nice to display multiple Bluetooth devices. At least 2, Mouse and Keyboard in two row format like Fan Speed for example.

exelban, 2021-08-16:
Hi. Yes, it could be implemented. But will remove the battery widget(

wfrank21, 2021-08-17:
Why would it remove the battery widget?

exelban, 2021-08-17:
It's related to the selector of which device to show. To show multiply devices - requires the option to select multiply devices. The battery widget needs one device. And I don't see a good way of how it could be realized.

exelban, 2021-08-22:
Hi. So, do I change the widgets?

wfrank21, 2021-08-23:
Why not have a selector in the Bluetooth Menu like in Fans?
[image]
[image]

exelban, 2021-08-23:
Yeap, it's exactly what must be done. But, the battery widget requires one value, and what if a few batteries are enabled. Which battery to show in the battery widget?

wfrank21, 2021-08-23:
It would be nice if it would be in the two row format like fans or sensors. Then prevent the user to enable more than two batteries.

exelban, 2021-08-23:
I'm not sure I understand.
To have a few batteries levels in the menu bar, you must have an option to enable a few batteries.
But the `Battery` widget could have only one value.

So, if you have enabled a few batteries from the list. How to choose which battery level to show in the `Battery` widget?

wfrank21, 2021-08-23:
Well if you can only have one value, the issue can be closed, because I wanted to see both values in two rows. BUT what can be done: Show the lower value, because the lower value is probably more important to the user.
If Mouse has 50% but Keyboard only 30%, show Keyboard because the user needs to charge it sooner.

exelban, 2021-08-23:
I suppose there is some misunderstanding.
There are 2 types of widgets:
- `Battery`<img width="57" alt="Zrzut ekranu 2021-08-23 o 12 43 56" src="https://user-images.githubusercontent.com/13332412/130434775-5b5cfcfe-5bee-42b7-877c-17ec82fe4e28.png">
- `Sensors` (one/two rows): [image]

The first one (Battery) requires only 1 value. The second one (Sensors) works with multiply values.
So, adding the `Sensors` widget to the battery module requires adding the option to select multiple values. But, in this case, the `Battery` widget (`Mini` as well) could not be presented in this module. Because these widgets (`Battery` and `Mini`) need only 1 value. But the module will have selected a few batteries, and it's impossible (without random) to show one value when a few present.

So. I could realize this feature and add the option to show multiply batteries in the menu bar. But in this case I must remove the `Battery` and `Mini` widgets from the battery module. And I'm not sure if it's a good transformation. 

I want to have a few batteries in the menu bar too, but also I like the `Battery` widget which shows my iPhone battery level. So it's hard to decide for me what option to choose(

wfrank21, 2021-08-23:
And how is the Bluetooth option displayed? Through `Battery`, `Sensors` or is there a dedicated `Bluetooth` Module?

exelban, 2021-08-23:
There is a dedicated module for Bluetooth.

wfrank21, 2021-08-23:
So as we (I) want to see two Bluetooth Battery levels, why bother with the Battery module and just use the Bluetooth Module instead? Or am I missing something here?

exelban, 2021-08-23:
The battery module is for MacBook battery stats. The Bluetooth module is for the Bluetooth device's battery level. 
Why are you asking for a Battery module? 
The future is related to the Bluetooth module I suppose.

wfrank21, 2021-08-24:
Because you told me that the `Battery` Module only works with one value, so I thought the Bluetooth Battery Levels are shown through the `Battery` Module

exelban, 2021-08-24:
I was talking that now you have the option to select only one battery to show in the Bluetooth module.
And to show multiply values, you need to have an option to select a few of them.

wfrank21, 2021-08-24:
So when we only can show one value at a time, is it possible to show the lowest value?

exelban, 2021-08-24:
No. Showing the lowest value is not correct. Because it will be hard to say which battery level I see in the menu bar at this moment: keyboard or mouse. It must be defined level.

I have decided what to do.
I will remove the `Mini` and `Battery` widgets from this module. And add a `Sensors` widget. It will allow showing any amount of batteries.

Let's keep the `Battery` widget reserved only for the battery module. And it will be less complicated with it.
It's also hard to say which battery you looking for: MacBook or AirPods (in the case when enabled both modules (battery and Bluetooth) with the selected `Battery` widget)

closed 2021-09-05 by exelban