exelban/stats #590: Feature Request: "Update Frequency with Panel Open"

Panini, 2021-08-17:
**Describe the bug**
**Update frequency** 1sec is a little cpu intensive, so I use higher values.
But sometimes I want to monitor something in real time (1sec), and by then I've usually clicked the menubar icon already and have the **panel open**.

- (for example I wanted to monitor DC-in wattage in real-time to see how quickly (if at all) it takes to reach max charging speed)

**Details:**
 - Device: [Mid 2017 15" Macbook Pro]
 - MacOS [10.15.7]
 - Application version [e.g. 2.6.5]

exelban, 2021-08-17:
Hi. Do you propose to have another option that will change the update interval when the popup window will be opened?

Panini, 2021-08-19:
> Hi. Do you propose to have another option that will change the update interval when the popup window will be opened?

Yup!

exelban, 2021-08-19:
Sorry but it's not possible. It will require rewriting at least half of the app.
And I don't think that it's a good feature to implement at all. Because it's such unexpected behavior. If CPU must update every second. It must update every second always. And not in some cases, it must update faster or slower.

closed 2021-08-19 by exelban