exelban/stats #591: Repeatedly crashing

richyliu, 2021-08-19:
**Describe the bug**
The app frequently crashes during normal usage. This only happened after I upgraded to version 2.6.5 of the app. I have also encountered this issue on MacOS 11.5.1. I have attached several crash logs below from `~/Library/Logs/DiagnosticReports` (I had to add the .txt suffix to upload to Github).

**Details:**
 - Device: Macbook Pro 2020 (M1)
 - MacOS: 11.5.2
 - Application

[Stats_2021-08-18-102017_Richard-MBP.crash.txt](https://github.com/exelban/stats/files/7010865/Stats_2021-08-18-102017_Richard-MBP.crash.txt)
[Stats_2021-08-18-113553_Richard-MBP.crash.txt](https://github.com/exelban/stats/files/7010866/Stats_2021-08-18-113553_Richard-MBP.crash.txt)
[Stats_2021-08-18-164123_Richard-MBP.crash.txt](https://github.com/exelban/stats/files/7010867/Stats_2021-08-18-164123_Richard-MBP.crash.txt)
 version 2.6.5

exelban, 2021-08-19:
Hi. Thanks for reporting.
Please disable the Bluetooth module, it causing the crash. I will try to figure out why it's happening.

richyliu, 2021-08-19:
Thank you for your quick reply. I will do that.

exelban, 2021-08-19:
I don't have an Apple Silicon mac, so I cannot reproduce the bug. Could I ask you to run a few builds and tell if the application crashes? (Please enable Bluetooth module for test, even if it does not have any device to show)

[Stats.dmg.zip](https://github.com/exelban/stats/files/7016648/Stats.dmg.zip)

jack24254029, 2021-08-23:
I received the crash too.
When my MacBook Pro wakeup, the Stats will be crash sometimes.

<details>
<summary> MackBook Info </summary>

- MacBook Pro 13" (Mid 2017)
- macOS Big Sur (11.4.0)

</details>

<details>
<summary>Crash Report</summary>
<p>

```
Process:               Stats [962]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.6.5 (296)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Stats [962]
User ID:               502

Date/Time:             2021-08-20 20:05:19.411 +0800
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        807565F1-C566-AB86-8224-A3DB19EB8FA0

Sleep/Wake UUID:       FC5400F4-BD3D-4750-B08B-F2E107FAE8BA

Time Awake Since Boot: 110000 seconds
Time Since Wake:       2400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [962]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   eu.exelban.Stats.Bluetooth    	0x000000010052f085 0x100521000 + 57477
1   eu.exelban.Stats.Bluetooth    	0x000000010052b6c4 0x100521000 + 42692
2   com.apple.CoreBluetooth       	0x00007fff308c5271 -[CBCentralManager handlePeripheralDiscovered:] + 1703
3   com.apple.CoreBluetooth       	0x00007fff308c7048 -[CBCentralManager handleMsg:args:] + 811
4   com.apple.CoreBluetooth       	0x00007fff308b9037 -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
5   com.apple.CoreBluetooth       	0x00007fff308b8f73 __30-[CBXpcConnection _handleMsg:]_block_invoke + 66
6   libdispatch.dylib             	0x00007fff201a7623 _dispatch_call_block_and_release + 12
7   libdispatch.dylib             	0x00007fff201a8806 _dispatch_client_callout + 8
8   libdispatch.dylib             	0x00007fff201ae5ea _dispatch_lane_serial_drain + 606
9   libdispatch.dylib             	0x00007fff201af0e0 _dispatch_lane_invoke + 417
10  libdispatch.dylib             	0x00007fff201b4a08 _dispatch_main_queue_callback_4CF + 613
11  com.apple.CoreFoundation      	0x00007fff204885e8 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
12  com.apple.CoreFoundation      	0x00007fff2044a7d2 __CFRunLoopRun + 2755
13  com.apple.CoreFoundation      	0x00007fff2044964c CFRunLoopRunSpecific + 563
14  com.apple.HIToolbox           	0x00007fff28687ab3 RunCurrentEventLoopInMode + 292
15  com.apple.HIToolbox           	0x00007fff28687815 ReceiveNextEventCommon + 587
16  com.apple.HIToolbox           	0x00007fff286875b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
17  com.apple.AppKit              	0x00007fff22c4a6f2 _DPSNextEvent + 864
18  com.apple.AppKit              	0x00007fff22c48ec5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
19  com.apple.AppKit              	0x00007fff22c3b239 -[NSApplication run] + 586
20  eu.exelban.Stats              	0x000000010044663b 0x100442000 + 17979
21  libdyld.dylib                 	0x00007fff2036df5d start + 1
[... 493 lines cut ...]
MALLOC guard page                   48K       11 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
SQLite page cache                  128K        2 
STACK GUARD                       56.0M        9 
Stack                             12.1M        9 
VM_ALLOCATE                         72K       16 
__DATA                            14.1M      362 
__DATA_CONST                      16.4M      213 
__DATA_DIRTY                      1164K      132 
__FONT_DATA                          4K        1 
__LINKEDIT                       500.8M       33 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2480K        2 
__TEXT                           242.4M      363 
__UNICODE                          588K        1 
libnetwork                         128K        8 
mapped file                      348.1M       47 
shared memory                      756K       15 
===========                     =======  ======= 
TOTAL                              1.8G     1697 
TOTAL, minus reserved VM space     1.4G     1697 
```

</p>
</details>

exelban, 2021-08-23:
@jack24254029 thanks for reporting. Looks like it's the same bug.

exelban, 2021-08-23:
Could someone check this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7032772/Stats.dmg.zip)

jack24254029, 2021-08-24:
> Could someone check this build?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/7032772/Stats.dmg.zip)

I can check this.
But I need a few days.

richyliu, 2021-08-25:
> Could someone check this build?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/7032772/Stats.dmg.zip)

I think this build is good. I have been running the app for a day now and it doesn't crash when I wake up my computer.

exelban, 2021-08-25:
@richyliu thanks for the check.

closed 2021-08-29 by exelban