exelban/stats #593: Option to hide the kill button

reekystive, 2021-08-23:
Stat is very easy to use. It is lightweight. I just use it as a simple monitor, and running it all the time.

In the most recent update ([v2.6.6](https://github.com/exelban/stats/releases/tag/v2.6.6)), I noticed that kill buttons are added in the dashboard (https://github.com/exelban/stats/issues/569). I think in such a monitor, the kill buttons should not be placed in such an obvious place that are easy to be touched because **it's very dangerous**.

I tried clicking the kill button, and it didn't even give any warning before the action. If the process belongs to root, nothing will happen when you click the button. In fact, when I decide to kill a process, I will choose to use the macOS Activity Monitor because it can provide more information before killing the process.

I suggest adding an option to let the user decide whether to show kill buttons, or just roll back that commit. However, the above are only suggestions. If you have different views, please feel free to put forward them!

I'm going back to the previous version now.

[image]

exelban, 2021-08-23:
Hi. Thanks for the feedback.
I agree with your points. I think it's too much for the monitor app. Also, it's very hard to kill the app you want. Because it could be replaced by another app when click on it (if the update interval is 1s). So, to keep this feature I will need to make some UX to "pin" the process by click or something like this. (what I don't want to do, because it's a little complicated)

I am more inclined to drop this feature from the app. (I prefer to use Activity Monitor or terminal to kill the app).

sddgit, 2021-08-24:
I agree. Probably best to not have it at all. Also, the top processes are now listed in reverse order by cpu usage, rather than top down.

exelban, 2021-08-24:
@sddgit yeap, I have noticed this bug too. Will be fixed.

closed 2021-09-05 by exelban

ericboehs, 2021-12-08:
It's too bad this was removed instead of implemented as requested.

If my machine is freezing up due to an unruly process, it's a lot to ask of it to open another program when system resources are tied up. Having a kill button a click away was nice.