exelban/stats #601: Feature request: M1 - bundle CPU temp sensors and show only highest (like "Hot" app)

ximmox, 2021-08-31:
Hi, 

I really love Stats and I would like to suggest a little feature that makes me use the "Hot" app (source and info: https://github.com/macmade/Hot) for showing the CPU-temperature currently instead of using Stats for it.

[image]

Hot reads out all the relevant CPU temp sensors on the M1, which seem to be the eACC and pACC MTU Temp Sensors with different numbers (like in the screenshot).
For the menubar icon, it only shows the highest temperature value of those, which is the most relevant info for the user I suppose.
I don't care which core is how hot currently, I would like to see with a glance what's the highest my overall CPU is at the moment.

Maybe it is possible to add some kind of "sensor bundling"? For example make a virtual sensor, which shows the highest temp of X chosen sensors you want to be monitored.
Or, more simply: add a sensor-entry like "show only highest CPU temp" in the sensors menu.

Please let me know if there is any more info that I can provide or if I can test anything on my MacBook Air.

Thanks!

exelban, 2021-08-31:
Duplicate of #473

closed 2021-08-31 by exelban

ximmox, 2021-08-31:
How stupid of me to miss that, sorry! I searched for issues with "temp", "temperature" and so on and #473 didn't come up.