exelban/stats #603: Temperature for ssd and hhd

notyz77, 2021-09-01:
I am using this app from months. This app is very helpful to see hardware usage. The only problem is that I can't see Temperature for ssd and hhd and also nvme drive life status. Can you add those feature.

exelban, 2021-09-01:
Hi. Thanks for the feedback.
HDD or SSD temperature you could see in the sensors module (if it's available).
NVMe life status, I'm not sure if it will be possible to add. I will check this. Or maybe do you know if there is any API in macOS for this?

notyz77, 2021-09-02:
still I can't see temperature for my nvme and HDD because I think it is a hackintosh system but their is app called HWMonitorSMC2 with that i can see temperature and nvme life status.
app link:https://github.com/CloverHackyColor/HWMonitorSMC2
[image]

exelban, 2021-09-05:
Hi.
There are only 2 options to fetch the NVME health: using `smartctl` or implement a SMART "driver" in the app. I would be like to use `smartctl`, but it does not preinstall on macOS. So, it could not be used. Writing own SMART "driver" is an option. But unfortunately, I will have no time to do it. I would be glad if someone would like to implement this feature via PR. 
For me, it's not useful information, and I don't see a reason to have it.

closed 2021-09-05 by exelban