exelban/stats #604: Menus overlapping

erentantekin, 2021-09-04:
**Describe the bug**
Not sure if possible via MacOS APIs, but it would be cool if stats could close existing open menus in the menu bar. Clicking a stats menu sometimes results in two menus overlapping as seen in the screenshot 

[image]

**Details:**
 - Device: 16 inch macbookpro
 - MacOS 11.5.2
 - Application version 2.6.0

exelban, 2021-09-05:
Hi. Thanks for the feedback.
Normally details must be closed if click outside the window. It's not fully controlled by API, but via macOS.

Does it happen with any apps? Or stats does not hide only when are you opening some specific app?

exelban, 2021-09-08:
Hi. Ok, I figured out when it happens.
It happens when the user opens the macOS popover window. Opening the popover window does not change the current app focus. As a result, Stats does not lose focus and doesn't know that it must close the window. It cannot be fixed. It's how macOS works(

closed 2021-09-08 by exelban