exelban/stats #605: No menubar icons (macOS Monterey)

tylergannon, 2021-09-05:
**Describe the bug**

Nothing appears in the menubar.  I'm fully aware that I'm running a beta version of the operating system so no pressure on you, but thought you might like to be prepared for possible breaking changes in menubar functionality with Monterey.

**Details:**

[image]

 - Application version 2.6.8

exelban, 2021-09-05:
Hi. Thanks for the feedback.
As you noticed, it's a beta version. I don't have a Monterey to check why the app does not work. 
But I still will wait for the release. (PS: I think it's more related to the Apple Silicon platform, not to the macOS).

Could you maybe check if you have a crash report of Stats? It could be found in `Applications -> Utilities -> Console`. Find the Crash Reports tab. There you must see a row called Stats.

tylergannon, 2021-09-06:
Oh, that's an interesting hunch, that it's more related to the architecture change.
I don't see any crash reports in Console.app.  Here's a little more information though:

I started the app a few times, both by clicking on the icon in Finder, as well as using the `open -a /Applications/Stats.app/` from a terminal window.  The result was the same each time.  No error message, the settings window will appear, but nothing in the menu bar.

Now, here's the magic!

I rebooted my computer this morning, after logging in, there's the info in the menu bar...

[image]

So maybe for the time being, it could be helpful to add advice into your README, that logging out / logging in can produce results for users who don't see changes immediately after installation.

exelban, 2021-09-06:
Hmm. Interesting, so I suppose it's related to some bug in the Monterey, not in the app. I will prepare the app before the public release of Monterey.

closed 2021-09-06 by exelban