exelban/stats #606: nettop high CPU usage

dima-xo, 2021-09-06:
**Describe the bug**
Nettop uses a lot of cpu.

Sample of stats:
[Sample of Stats.txt](https://github.com/exelban/stats/files/7113633/Sample.of.Stats.txt)

Parent process — stats.
[image]
[image]

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - MacOS [e.g. 10.15.5]
 - Application version [e.g. 2.1.11]

dima-xo, 2021-09-06:
2.6.8 same results

exelban, 2021-09-06:
Hi. Thanks for the reporting.
`nettop` must running only when you have opened the `Network` details window. Does the nettop run always (even when you do not have any Stats window opened)?

exelban, 2021-09-06:
Could you please check what `Reader type` you have selected in the `Network` module settings?

dima-xo, 2021-09-06:
> Hi. Thanks for the reporting.
> `nettop` must running only when you have opened the `Network` details window. Does the nettop run always (even when you do not have any Stats window opened)?

Yes, unfortunately. It is always launched, sometimes 2-3 instances at a time.
When minimized, the statistics do not work. Always shows up: 0, down: 0.

> Could you please check what Reader type you have selected in the Network module settings?
Process based. If I choose "interface based" with or without autodetection, the same result — nettop doesn't unload from memory.

(I'm using a VPN through wireguard if that helps, but before (a few versions ago, I don't remember when exactly) everything was stable)

exelban, 2021-09-06:
Hmm, so even when you have selected `interface based` the `nettop` still runnning?
(VPN do not have an impact, it could only provide wrong stats)

dima-xo, 2021-09-06:
Yes, I can describe the case:

Offload the nettop from memory (via force quit).
I enable Network in stats with interface based / process based.
Nettop is not running.
I click on network on top bar for details.
Nettop is up and running.
I hide the network's details.
nettop stays in memory.

It's worth noting that when Interface based is enabled, the stats on the top panel are updating (when Process based is enabled, zeros are displayed) 

nettop is triggered by a detailed statistics of the network. But it doesn't offloaded from memory.

dima-xo, 2021-09-06:
If I choose "number of top process" to 0, nettop isn't triggered by detail statistics tap.

exelban, 2021-09-06:
Thanks for the details. I will take a look at this strange behavior later today.

exelban, 2021-09-19:
Hi. Unfortunately, I cannot say why it's happening. 
I cannot reproduce this bug on intel based mac. It looks like it appears only on apple silicon based macs.

Brabalawuka, 2021-10-08:
On MacBook Pro 16. The nettop process somehow gets invoked and never stops even if i close the top bar deatils in network session

- v2.6.13

[image]

[image]

legzo, 2021-11-23:
Same behaviour on my machine : 
[image]

exelban, 2021-11-25:
Hi. Could please someone run `nettop -P -L 1 -k time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch`.
After that check if the `nettop` still in the system by `ps aux | grep nettop`.

legzo, 2021-11-30:
> Hi. Could please someone run `nettop -P -L 1 -k time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch`. After that check if the `nettop` is still in the system by `ps aux | grep nettop`.

Hi Serhiy,

It seems that nettop is not present after running the command :

```bash
~ 
→ nettop -P -L 1 -k time,interface,state,rx_dupe,rx_ooo,re-tx,rtt_avg,rcvsize,tx_win,tc_class,tc_mgt,cc_algo,P,C,R,W,arch
,bytes_in,bytes_out,
kernel_task.0,172339,10888,
vpnagentd.71,0,0,
macmnsvc.79,0,0,
masvc.80,6715,12670,
remoted.85,30760,37372,
apsd.113,5617,6432,
syspolicyd.148,86819,31377,
mDNSResponder.256,383201,82567,
softwareupdated.284,0,0,
com.cisco.anyco.288,479,87,
SubmitDiagInfo.302,750,180,
mobileactivatio.466,5464,7008,
biometrickitd.525,2246,1970,
ControlCenter.660,0,0,
rapportd.680,0,0,
CalendarAgent.691,0,0,
corespeechd.742,374,305,
parsecd.785,1312,1034,
Google Chrome H.1047,841972,67725,
AssetCacheLocat.1132,12353,5863,
firefox.6238,2842249,1170451,
.21176,368,368,
VShieldScanner.93560,7221,4607,
VShieldScanner.93561,6063,2491,
VShieldScanner.93562,6834,3907,
Google Drive.17418,18752,203844,
Google Drive He.17434,2808,2086,
AddressBookSour.50268,0,0,

~ 
→ ps aux | grep nettop
jterrier         51766   0.0  0.0 34253844    900 s001  S+    7:33PM   0:00.00 grep --color=auto nettop
```

exelban, 2021-11-30:
Thanks. I could make a few debug versions with disabled different function which uses `nettop`. It allows figuring out what exactly call makes a leak. But someone will need to run them and give feedback. Give me know if someone would like to help with this bug.

dima-xo, 2021-11-30:
I have no longer have this error reproducing, the nettop process is no longer present in activity monitor also. 
stats: V2.6.21
macOS 11.6, m1

exelban, 2021-12-01:
Hi.
I suppose it's some bug in the nettop or macos. Because not in the app for sure. I tracking nettop usage for some time, and don't found any issue with it. Additionally, the Stats does not run `nettop` in the loop, it only runs one read. And there is no chance that Stats does not close it. It must be closed by themself. It could be running only if the nettop does not stop after executing. But it must. Unfortunately, I cannot do anything with this. There are no alternatives that I can use for fetching the network processes.

closed 2021-12-01 by exelban

MORECATS, 2022-01-30:
Hello, in my observation, if you run nettop from a sandbox app, it will cause a very high cpu usage, if you simply disable sandbox and run again, cpu usage back to normal. It seems relevant to the sandbox, I'm still looking for a solution.

exelban, 2022-01-30:
@MORECATS thanks. But Stats is not sandboxed.

chrisgrieser, 2022-02-17:
can report that occasionally, nettop memory usage gets really high. requiring to kill the process to get stuff back to normal. I identified that it is Stats creating the process via activity monitor

perphilip, 2022-05-12:
I experienced this today as well, 2-3 nettop processes at ~138% cpu usage right after fresh install from Brew. Version 2.7.17. Only way to stop fans from going 100% was to terminate the processes.

iain17, 2022-08-22:
Still a problem on version 2.7.31. Not sure why this issue is closed?

heyheyhello, 2023-01-09:
This is still an issue, but interestingly if I kill Stats.app (nicely via settings > power button icon) then nettop is still left as an orphaned process? Maybe it's adopted by launchd. Notably this means we can't "Just turn it off and on again to fix it" :/

[image]