exelban/stats #607: sensor menubar item count is increasing

emindeniz99, 2021-09-06:
**Describe the bug**
[image]
for each update of sensors, it adds new value to the menubar

**Details:**
 - Device: Macbook Air M1
 - MacOS: 11.5.2
 - Application version: 2.6.8

exelban, 2021-09-06:
Hi. Thanks for the feedback.
Does it appear only in the new update?

emindeniz99, 2021-09-06:
> Hi. Thanks for the feedback.
> Does it appear only in the last update?

Hi, yes, it appears after the update to 2.6.8 from 2.6.7.

exelban, 2021-09-06:
yeap, there is a bug(
I will fix it today and release a new version with this fix.

exelban, 2021-09-06:
Could I ask you to try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7116870/Stats.dmg.zip)

emindeniz99, 2021-09-06:
> Could I ask you to try this build?
> [Stats.dmg.zip](https://github.com/exelban/stats/files/7116870/Stats.dmg.zip)

I tried, It works

exelban, 2021-09-06:
Thanks for reporting. I have made a [new release](https://github.com/exelban/stats/releases/tag/v2.6.9) with this hotfix.

closed 2021-09-06 by exelban