exelban/stats #608: High memory usage

rbrtmrtn, 2021-09-06:
Thank you for your hard work on this project, it seems like a great tool :)

**Describe the bug**

Stats is using > 1.3 GB of RAM on my machine. Fresh install via `brew` just about an hour ago.

**Details:**
 - Device: MacBook Air (M1, 2020)
 - MacOS: Big Sur 11.5.1
 - Application version: 2.6.8

[image]

exelban, 2021-09-06:
Hi. Thanks for the feedback.
Does it appear only in the new update?

rbrtmrtn, 2021-09-06:
Hi @exelban, thank you for the quick response. I'm not sure what you mean by new update—I just installed Stats for the first time tonight.

exelban, 2021-09-06:
Could you maybe try to install a [previous version](https://github.com/exelban/stats/releases/tag/v2.6.7) and check it?

rbrtmrtn, 2021-09-06:
Is there a way to do that via Homebrew? I'm happy to try an earlier version, but I try to use a package manager to the extent possible.

exelban, 2021-09-06:
Unfortunately I'm not an active user of brew, and don't know if there is an option to install a specific version of the app via cask(

rbrtmrtn, 2021-09-06:
No worries! I installed 2.6.7 manually and this version doesn't seem to have the memory issue. ✅

(Also, before uninstalling 2.6.8, I noticed it was using quite a bit of my CPU, too; 2.6.7 is normal so far.)

exelban, 2021-09-06:
Thanks. So please use `2.6.7` till I will fix it. Looks like this issue is connected in some way with #606

rbrtmrtn, 2021-09-06:
Got it, thank you @exelban! Let me know if I can help in any way 😄

exelban, 2021-09-06:
Could I ask you to try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7116871/Stats.dmg.zip)

rbrtmrtn, 2021-09-06:
@exelban Sure thing, I just installed it and will report back after it's run for a while.

rbrtmrtn, 2021-09-06:
@exelban It's been running for a little under 20 minutes and it's only using 35 MB of RAM. I think that fixed it! 🎉

exelban, 2021-09-06:
Thanks for reporting. I have made a [new release](https://github.com/exelban/stats/releases/tag/v2.6.9) with this hotfix.

closed 2021-09-06 by exelban

rbrtmrtn, 2021-09-06:
Awesome, many thanks for addressing this so quickly @exelban! I created a PR over at Homebrew to update the cask to 2.6.9.