exelban/stats #610: Bug? Average CPU Temp gets stuck after some time/not updating

ximmox, 2021-09-06:
Hi, 
with the new version 2.6.8 I am using the new average CPU and average SOC Temp sensors in the menubar.
After sleep/opening my Macbook Air M1 this morning, the values in the menubar and the sensors dropdown don't change for the average values. The other "real" sensors do change:

[image]

Closing/reopening Stats does mitigate this.
Also the high memory usage described in #608 drops from around 750MB before closing to around 66MB after restart for Stats.

Apart from this bug I would like to discuss something else related to the Average temperature sensors:
In my opinion it would be better to show the highest temp reading of all the sensors instead of the average, as the maximum temperature of any part/core of the CPU is what I'm more interested in than an average (similar to how the "Hot" app does it).

Thoughts?

Thanks, immo

exelban, 2021-09-06:
Could I ask you to try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/7116872/Stats.dmg.zip)

ximmox, 2021-09-06:
Sure, thank you, running it now.

To see if the memory leak is still there I can report back in an hour or two, for the temperatures not updating in the menubar I might need to wait for a longer sleep period / tomorrow morning (7 PM in Germany right now).

rajtslegr, 2021-09-06:
So far so good for me. We will see.

ximmox, 2021-09-06:
Same here, memory usage went from 38MB after fresh start to 53MB currently, which is totally fine.

Average temperatures for CPU and SOC in menubar are still updating as expected.

exelban, 2021-09-06:
Now about an average value.
I would like to leave an average of all values. Because `average` means an average value of all cores, not the hottest value.

But I'm ready to add for example `Hottest CPU core` etc. What do you think?

ximmox, 2021-09-07:
Result after a night's sleep: values still updating in the menubar, memory usage 54MB.
Seems fixed, that's great.

Yeah of course, If you would like to keep the average values and don't mind adding Hottest CPU, GPU and SOC, that would be even better for all users.

closed 2021-09-12 by exelban

Idomo, 2021-12-15:
Hi @exelban, I have MBP 16" with M1 Pro macOS 12.1 and Stats 2.6.23.
I can't find the Average/Hottest options in the Sensors menu, Do you aware to this issue?

exelban, 2021-12-15:
@Idomo hi, please check opened issues.