exelban/stats #614: [Apple M1] Memory usage difference

livesamarthgupta, 2021-09-09:
The activity monitor shows different amounts of memory used while the app shows different amounts.

**Details:**
 - Device: Macbook Air M1
 - macOS: Big Sur 11.5.2
 - Application version: 2.6.9

exelban, 2021-09-09:
Hi. Could you provide more information?

livesamarthgupta, 2021-09-09:
[image]

livesamarthgupta, 2021-09-09:
You can see in the screenshot, the memory used in the activity monitor is 6.27GBs while the same in the app is 5.64GBs.

exelban, 2021-09-09:
Stats: 3.09 + 1.16 + 1.39 ~= 5.64
Activity Monitor: 3.09 + 1.15 + 1.39 ~= 6.27

You could find the explanation in Google or there was some issue with the explanation why Activity Monitor shows more.

closed 2021-09-09 by exelban

livesamarthgupta, 2021-09-09:
I know it's weird how the activity monitor is showing the used memory, I didn't see these in earlier mac models. I guess the memory management for M1 macs is somewhat different from earlier models.

exelban, 2021-09-09:
Nope, the same on intel-based.