exelban/stats #619: [Feature Request]: Custom widgets for Mac OS Big Sur

livesamarthgupta, 2021-09-12:
It would be good if this gets custom widgets to show stats directly from the widgets section.

exelban, 2021-09-14:
Hi.
What do you mean by `custom widgets`?

livesamarthgupta, 2021-09-15:
[image]
I mean something like this for the notification center.

exelban, 2021-09-15:
Hi. Oh, I see.
It's a really nice feature. But I will have no time to implement this. It will require creating new views for each module.
This app is more focused on the menu bar. So I cannot promise that it will be implemented.

exelban, 2021-09-17:
Hi. I suppose this feature could be implemented after #623.
But it will be next year (or maybe even later).

closed 2021-09-17 by exelban