exelban/stats #623: [Feature Request] A way to combine all the monitoring modules into one menubar icon.

DaMiggs, 2021-09-14:
I think this could be very helpful in clearing up menubar clutter.
I did see a photo of some feature for this app but when I tried to find it I couldn't.

exelban, 2021-09-17:
Hi. Thanks for propose.
There was already a ticket with this feature.
This feature requires a combined view for all widgets, which does not exist. And I'm not sure that it will be a good solution for this. Because it's a little bit hard to decide what to show in this view. And providing options where users can choose what to show will make too complicated settings.

I do not say that this feature will not appear in the app. But for sure it will be not this year. There are a lot of features that must be implemented before adding new functionality to the app.

closed 2021-09-17 by exelban

ericboehs, 2021-12-08:
I too would like to see a combined option as I really just want 1 icon for Stats in my menu bar but I'd love to see all the information Stats offers.