exelban/stats #628: Stats app keeps crashing on MacOS

harpreetsd9, 2021-09-16:
The app keeps crashing on MacOS version 11.6 (20G165) with no errors or messages.

exelban, 2021-09-16:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in `Applications -> Utilities -> Console`. Find the Crash Reports tab. There you must see a row called Stats.

harpreetsd9, 2021-09-16:
[stats.zip](https://github.com/exelban/stats/files/7180620/stats.zip)

rswail, 2021-09-17:
[stats-crashlogs.zip](https://github.com/exelban/stats/files/7182976/stats-crashlogs.zip)
Attached 5 logs, all showing SEGVs at various places/threads but all have Stats.Bluetooth in the traceback of the crashed thread.

exelban, 2021-09-17:
Thanks for sharing the crash reports. Yes, crashing is related to the Bluetooth module. Disabling the Bluetooth module will prevent the crash. 
I will try to fix it for the next release.

rswail, 2021-09-18:
No problem, it's not something I actually had displaying on the menu bar, but it was enabled as a module, if that helps the fix. Stats is stable now over 24+ hours, which is more than before :) Thanks again for the app

closed 2021-09-19 by exelban