exelban/stats #630: [Feature Request] Log historical data for sensors to provide graphs

jvacek, 2021-09-17:
There's currently no way to compare the values over time, would be sweet to have the option to log the temperatures and other sensor info. Given how many there are, I would also suggest providing a separate interval setting for the historical data there possibly? I'm sure there's plenty of other cleverer options for intelligently storing this.

iStats Menu did it quite well, and it's a feature I'm sorely missing. Then again, they only worked with _one_ sensor on the m1.

exelban, 2021-09-17:
Hi. Thanks for your feedback.
I suppose you propose to add the graph to sensors. It could be implemented, but it will take some time.

jvacek, 2021-09-22:
Yes, that's correct :)

exelban, 2021-09-22:
Yeap, it could be added.

laoyaoer, 2022-01-05:
Would be sweet if extend the usage history to 1 day, 1 week , 1 month.

artstorm, 2022-01-12:
Accessing historical data would be a great addition. I'd like to suggest an addition to this idea, to make the data accessible in a structured format, so it can be collected, stored and visualized externally. One approach could be to add a command line option that just outputs the values of interest in a structured format.

I currently have this set up with iStats Menu so I pull all data into a Time Series database and then I visualize it with [Grafana](https://github.com/grafana/grafana). Doing it this way I can consolidate the relevant data and also have alerts for my different Macs and remote servers in one dashboard, accessible anywhere. And I can see years of history.

iStats Menu doesn't have this built in, so I do it by having my data collector parsing the iStats sqlite database on a regular interval, and then reformat the values from the DB to a Time Series compatible format. Ie, I've made a simple command line tool that reads iStats' data file and then outputs the data from iStats DB formatted like this:

```
thermal,host=Johans-MBP cpu=60.66796875,fan=1844.44189453125,gpu=61.0 1641960371000000000
thermal,host=Johans-MBP cpu=61.69140625,fan=1840.86480712891,gpu=61.0 1641960365000000000
thermal,host=Johans-MBP cpu=62.49609375,fan=1837.28771972656,gpu=61.0 1641960359000000000
thermal,host=Johans-MBP cpu=62.40625,fan=1828.31823730469,gpu=61.0 1641960353000000000
thermal,host=Johans-MBP cpu=63.5625,fan=1819.34875488281,gpu=61.0 1641960347000000000
thermal,host=Johans-MBP cpu=62.0859375,fan=1826.59149169922,gpu=62.0 1641960341000000000
thermal,host=Johans-MBP cpu=64.84765625,fan=1833.83422851563,gpu=63.0 1641960335000000000
thermal,host=Johans-MBP cpu=66.02734375,fan=1831.42114257813,gpu=61.0 1641960329000000000
thermal,host=Johans-MBP cpu=68.98046875,fan=1829.00805664063,gpu=59.0 1641960323000000000
thermal,host=Johans-MBP cpu=59.3984375,fan=1831.38842773438,gpu=58.5 1641960317000000000
thermal,host=Johans-MBP cpu=58.03125,fan=1833.76879882813,gpu=58.0 1641960311000000000
```
Ie, a name, a machine id, the metrics I'm interested in, and a timestamp.

In my setup, I use [InfluxDB](https://github.com/influxdata/influxdb)  for time series data storage and then visualized with a Grafana dashboard where all my different machines' data are available.

Would this be a feature to consider? To be able to access the Stats data in a structured format? For instance via command line or writing to a log file that can be parsed. I believe all different Time Series DBs can parse the [Prometheus Data Model](https://prometheus.io/docs/concepts/data_model/), JSON would be another option that should be parseable by any data collector

exelban, 2022-01-17:
@artstorm hi. There is no chance that Stats will write the data to some normal db (time series or even document). It's required to have this db on the machine. Because (as I know) there is no influx db preinstalled on macos. The only way to store the data it's some sort of SQLite db.

closed 2022-03-06 by exelban