exelban/stats #636: fan setting is not working after the awake

erkanzileli, 2021-09-19:
**Describe the bug**
1. Set fans to work at the most speed
2. Let the PC sleep. The fans have to stop.
3. Wake up the PC
4. Fans are working on the Automatic mode but the Fan menu isn't say so

Here is an image from after awaking

[image]

**Details:**
 - Device: [e.g. Macbook Pro 16" 2019]
 - macOS: [e.g. 11.5.2]
 - Application version: [e.g. 2.6.6]

exelban, 2021-09-19:
Hi. Thanks for reporting.
Yes, macOS resets the fans after sleep. I will add the listener for awake from sleep to check fans mode.

ScizorsBLBC, 2021-10-06:
How do you view the fan speed mine doesn't show any fan info in the settings?