exelban/stats #645: feature request/enhancement - ability to disable turbo boost

the-woocash, 2021-09-26:
Since on MBP 16" running an external display always enables discrete GPU, it is helpful to disable turbo boost to help with heat problems and make fans spin slightly slower and quieter.
I am aware that i can manually control fans, but maybe it would be possible to incorporate a toggle for turboboost? 
For now i am using this (https://github.com/rugarciap/Turbo-Boost-Switcher), maybe it would be possible to incorporate the functionality in Stats?

exelban, 2021-09-27:
Hi. Thanks for the proposal.
I'm not sure that I will add this feature. Because this feature is more about control your mac. And Stats is more about monitoring. `Turbo-boost-switcher` is a nice app. And if you need to disable turbo boost I propose you use this app.
Also disabling turbo boost is not a simple process. And I'm not sure I will find time to debug how it works and could be disabled.

PS: it could be added if turbo boost could be easily toggled by SMC, but someone must check it

closed 2021-09-28 by exelban

vladsboev-web, 2026-10-27:
For anyone landing here via Google who wants lightweight CPU/GPU thermal monitoring combined with a 1-click Intel Turbo Boost toggle and fan control, take a look at **TurboControl**: https://github.com/vladsboev-web/TurboControl

It's an open-source, native Menu Bar app for Intel Macs (macOS Monterey–Sequoia) designed specifically to address the MBP 16" heat/throttling issue without password prompts.