exelban/stats #649: #BUG  menu bar failure of sensor information

StevenLi-phoenix, 2021-09-27:
When I reboot my mac. The sensor information did not display on menu bar of my mac.

exelban, 2021-09-27:
Hi. Thanks for reporting. Could you please share the crash report?
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

Also, please fill the issue according to the template. Maybe you running an app on hackintosh and it does not relate to the app.

StevenLi-phoenix, 2021-09-27:
[image]
This is what i display When I reboot the system.
With no sign of SEN option on menu bar.

StevenLi-phoenix, 2021-09-27:
Which I am sure that I open it while I close my mac.

StevenLi-phoenix, 2021-09-27:
[Stats_2021-09-23-174231_lishuyudeMacBook-Pro.crash.zip](https://github.com/exelban/stats/files/7236087/Stats_2021-09-23-174231_lishuyudeMacBook-Pro.crash.zip
[Stats_2021-09-21-123706_lishuyudeMacBook-Pro.crash.zip](https://github.com/exelban/stats/files/7236091/Stats_2021-09-21-123706_lishuyudeMacBook-Pro.crash.zip)
)

StevenLi-phoenix, 2021-09-27:
It is not a crash problem. This issue occur whenever I reboot my mac.

This is what I want the meun bar look like(with sensor option)
[image]

Also, I am not using hackintosh. I am using 
MacOS Big Sur 11.6
MacBook Pro (13-inch, 2020, Four Thunderbolt 3 ports)
2 GHz 四核Intel Core i5
16 GB 3733 MHz LPDDR4X
Intel Iris Plus Graphics 1536 MB

StevenLi-phoenix, 2021-09-27:
[image]
There are only two crash report and I had already upload them. It seems it is not related to my issue.
The sensor meun bar just auto close when rebooting.

StevenLi-phoenix, 2021-09-27:
[crash_report.zip](https://github.com/exelban/stats/files/7236112/crash_report.zip)

exelban, 2021-09-27:
Thanks. Thanks for sharing.
Just to clarify: the app does not start after reboot or the app stops working when mac wake up from sleep mode?

StevenLi-phoenix, 2021-09-27:
No!
What I mean is the app start when login. Other option start successfully but sensor option closed automatically. And did not start. It do not display sensor on meun bar when reboot. But other option like CPU or GPU display success.

exelban, 2021-09-27:
Oh. Ok, now I see.
So, after restart `Sensors` module is enabled or disabled?

StevenLi-phoenix, 2021-09-27:
DISable

exelban, 2021-09-27:
And it was enabled before the restart, yes?

StevenLi-phoenix, 2021-09-27:
[image]
This touch bar is disable.

Yes. for latest question.

StevenLi-phoenix, 2021-09-27:
Yes, it is enable before reboot.

exelban, 2021-09-27:
Ok, thanks.
I will take a look at this.

StevenLi-phoenix, 2021-09-27:
Also the GPU model gives me my GPU is running at 280% when I am playing game with it. I am a little bit confuse but not treat it as an issue. I am just putting it here. 

Intel Iris Plus Graphics 1536 MB 

for my macbook 2020

[image]

exelban, 2021-09-27:
@StevenLi-phoenix do you have enabled at least 1 sensor?

exelban, 2021-09-27:
To GPU, could you please share the output of `ioreg -r -d 1 -w 0 -c "IOAccelerator"`?

StevenLi-phoenix, 2021-09-27:
I am sorry. Cuz I went to sleep.

Output if that:

+-o IntelAccelerator  <class IntelAccelerator, id 0x100000412, registered, matched, active, busy 0 (14 ms), retain 106>
    {
      "IOGeneralInterest" = "IOCommand is not serializable"
      "IODVDBundleName" = "AppleIntelICLGraphicsVADriver"
      "IOGVAHEVCEncode" = "2"
      "InternalStatisticsAccm" = {}
      "GPURawCounterBundleName" = "AppleIntelICLGraphicsMTLDriver"
      "MetalStatisticsName" = ("MetalStatistics","autocorr1/ComputeBasic","autocorr1/ComputeExtended","autocorr1/ComputeL3Cache","autocorr1/HDCAndSF","autocorr1/L3_1","autocorr1/L3_2","autocorr1/L3_3","autocorr1/RasterizerAndPixelBackend","autocorr1/RenderBasic","autocorr1/RenderPipeProfile","autocorr1/TDL_1","autocorr1/TDL_2")
      "IOGVAXDecode" = "2"
      "GPURawCounterPluginClassName" = "IGGPURawCounterSourceGroup"
      "InternalStatistics" = {}
      "IOGVABGRAEnc" = "Gen11"
      "AGCInfo" = {"fBusyCount"=0,"poweredOffByAGC"=No,"fSubmissionsSinceLastCheck"=9412,"fLastSubmissionPID"=137}
      "MetalPluginName" = "AppleIntelICLGraphicsMTLDriver"
      "PerformanceStatisticsAccum" = {"orphanedReusableVidMemoryHitRate"=0,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=11,"dataBufferCount"=0,"oolTexturePageInBytes"=0,"textureCount"=1710,"iosurfaceTextureCreationCount"=0,"surfaceReadLockIdleWaitTime"=0,"freeDataBufferWaitTime"=0,"orphanedReusableSysMemoryBytes"=317657088,"ioSurfaceReadOutBytes"=0,"agprefTextureCreationCount"=0,"orphanedReusableSysMemoryCount"=137,"finishAll2DWaitTime"=0,"orphanedNonReusableSysMemoryCount"=0,"ioSurfacePageOutBytes"=0,"surfaceBufferReadOutBytes"=0,"surfaceTextureCreationCount"=0,"textureReadOutBytes"=0,"gartMapInBytesPerSample"=0,"swapBytesPerSample"=0,"dataBytesPerSample"=0,"surfaceCount"=19,"clientGLWaitTime"=0,"surfaceCopyInWaitTime"=0,"clientSharedAllocatedBytes"=839680,"surfaceCopyOutWaitTime"=0,"gartMapOutBytesPerSample"=0,"ioSurfacePageInBytes"=0,"gartCacheBytes"=134217728,"volatileSurfaceCount"=0,"vramEvictionWaitTime"=0,"oolTextureCreationBytes"=0,"freeToAllocGPUAddressWaitTime"=0,"oolTextureCreationCount"=0,"orphanedReusableVidMemoryCount"=0,"orphanedReusableVidMemoryBytes"=0,"context2DCount"=3,"inUseSysMemoryBytes"=195330048,"bufferSwapCount"=0,"orphanedNonReusableVidMemoryBytes"=0,"contextGLCount"=3,"finish2DWaitTime"=0,"gartSizeBytes"=7516192768,"recoveryCount"=0,"stdTexturePageInBytes"=0,"agprefTextureCreationBytes"=0,"gartUsedBytes"=512987136,"surfaceWriteLockIdleWaitTime"=0,"surfaceBufferPageInBytes"=0,"finishGLWaitTime"=0,"iosurfaceTextureCreationBytes"=0,"finishCLWaitTime"=0,"finishVideoWaitTime"=0,"contextCLCount"=0,"freeSurfaceBackingWaitTime"=0,"swapCompleteGLWaitTime"=0,"orphanedNonReusableSysMemoryBytes"=0,"inUseVidMemoryBytes"=0,"agpTextureCreationBytes"=0,"stdTextureCreationBytes"=0,"agpTextureCreationCount"=0,"stdTextureCreationCount"=0,"surfaceSetShapeIdleWaitTime"=0,"gartFreeBytes"=7003205632,"textureVolunteerUnloadBytes"=0,"bufferFlipCount"=0,"hardwareSubmitWaitTime"=0,"swapComplete2DWaitTime"=0,"surfaceBufferPageOutBytes"=0,"surfaceBufferTextureCreationCount"=0,"hardwareWaitTime"=0,"contextVideoCount"=3,"freeSurfaceSwapBufferWaitTime"=0,"orphanedNonReusableVidMemoryCount"=0}
      "PerformanceStatistics" = {"orphanedReusableVidMemoryHitRate"=0,"texturePageOutBytes"=0,"swapCompleteVideoWaitTime"=0,"orphanedReusableSysMemoryHitRate"=11,"dataBufferCount"=0,"oolTexturePageInBytes"=5791744,"textureCount"=1710,"Device Utilization % at cur p-state"=3,"iosurfaceTextureCreationCount"=0,"freeDataBufferWaitTime"=0,"surfaceReadLockIdleWaitTime"=0,"orphanedReusableSysMemoryBytes"=317657088,"ioSurfaceReadOutBytes"=0,"agprefTextureCreationCount"=0,"orphanedReusableSysMemoryCount"=137,"finishAll2DWaitTime"=0,"orphanedNonReusableSysMemoryCount"=0,"ioSurfacePageOutBytes"=0,"surfaceBufferReadOutBytes"=0,"surfaceTextureCreationCount"=0,"textureReadOutBytes"=0,"gartMapInBytesPerSample"=0,"swapBytesPerSample"=0,"dataBytesPerSample"=0,"surfaceCount"=19,"Device Unit 0 Utilization %"=3,"clientGLWaitTime"=0,"clientSharedAllocatedBytes"=839680,"surfaceCopyInWaitTime"=0,"gartMapOutBytesPerSample"=0,"ioSurfacePageInBytes"=0,"surfaceCopyOutWaitTime"=0,"gartCacheBytes"=134217728,"volatileSurfaceCount"=0,"vramEvictionWaitTime"=0,"oolTextureCreationBytes"=5791744,"freeToAllocGPUAddressWaitTime"=0,"oolTextureCreationCount"=10,"Device Unit 1 Utilization %"=0,"orphanedReusableVidMemoryCount"=0,"orphanedReusableVidMemoryBytes"=0,"context2DCount"=3,"inUseSysMemoryBytes"=195330048,"bufferSwapCount"=0,"orphanedNonReusableVidMemoryBytes"=0,"contextGLCount"=3,"finish2DWaitTime"=0,"Device Unit 2 Utilization %"=0,"gartSizeBytes"=7516192768,"recoveryCount"=0,"stdTexturePageInBytes"=0,"agprefTextureCreationBytes"=0,"gartUsedBytes"=512987136,"surfaceWriteLockIdleWaitTime"=0,"surfaceBufferPageInBytes"=0,"finishGLWaitTime"=0,"iosurfaceTextureCreationBytes"=0,"finishCLWaitTime"=0,"finishVideoWaitTime"=0,"contextCLCount"=0,"Device Unit 3 Utilization %"=0,"freeSurfaceBackingWaitTime"=0,"swapCompleteGLWaitTime"=0,"orphanedNonReusableSysMemoryBytes"=0,"inUseVidMemoryBytes"=0,"agpTextureCreationBytes"=6119424,"stdTextureCreationBytes"=0,"agpTextureCreationCount"=16,"stdTextureCreationCount"=0,"surfaceSetShapeIdleWaitTime"=0,"gartFreeBytes"=7003205632,"textureVolunteerUnloadBytes"=0,"Device Unit 4 Utilization %"=0,"bufferFlipCount"=0,"hardwareSubmitWaitTime"=0,"surfaceBufferPageOutBytes"=0,"Device Utilization %"=11,"hardwareWai [... cut ...] ReportGroupName"="Internal Statistics","IOReportChannelInfo"={"IOReportChannelUnit"=0}})
      "IOMatchCategory" = "IOAccelerator"
      "AccelCaps" = 15
      "IOAccelDisplayPipeCapabilities" = {"DisplayPipeSupported"=Yes,"TransactionsSupported"=Yes}
      "IOGVAHEVCEncodeCapabilities" = {"VTPerProfileDetails"={"1"={"VTMaxEncodeLevel"=183},"2"={"VTMaxEncodeLevel"=183}},"VTQualityRating"=80,"VTRating"=100,"VTSupportedProfileArray"=(1,2)}
      "IOProbeScore" = 1000
      "SurfaceList" = ()
    }

StevenLi-phoenix, 2021-09-27:
> @StevenLi-phoenix do you have enabled at least 1 sensor?

No, maybe that is the problem.

exelban, 2021-09-28:
Very strange. But intel driver reports `"Device Utilization %"=308` in the report that you share.
I will limit that programatically.

closed 2021-10-03 by exelban