exelban/stats #687: "NET" label is added as the leftmost item or inconsistently

sbotnickusgs, 2021-11-05:
When enabling display of "NET" label for the network monitor when only the numeric display is selected, the label always displays to the far most left of all displays, not next to the network display which is to the right of some other performance displays. 
There is a workaround: add the graph display for network,  add the NET label(not pictogram), add the numeric, then remove the graph if you don't want it.   Playing around enabling and disabling the network displays and label yields variable results.

Macbook Pro, Big Sur 11.6 (Enterprise not permitting 11.6.1 yet)

Thanks for the useful app. I learned about it from Jeff Geerling on YouTube.

exelban, 2021-11-05:
Hi. Thanks for the feedback.
Sorry, but I totally do not understand what are you talking about. I have checked the Label widget and everything works fine.
Could you please share some screenshots of the bug?

sbotnickusgs, 2021-11-05:
It appears the character of what I am seeing is a bit different than what I initially described, it appears the order of adding items to view is important, so that is not a bug. What is happening is the order of data and label items is not persisting through login/logout.  So, what I am seeing is that if I add a label and data display for a given source in that order, after a login/logout the order will change to data and label. Perhaps just a feature, but it isn't consistent with the label behavior of something like the CPU bar graph where selecting the label option puts the text label to the left of the data display. Hopefully this and/or the screen shots make sense.

I have since installed 11.6.1
[image]
[image]

exelban, 2021-11-05:
And why you don't just order the items as you want?
If the order does not set up manually, the items in the menu bar appear in the module load order. So it's a little bit random.

sbotnickusgs, 2021-11-09:
The data displays do order as desired, but the label for them always goes to the right of the data display for a given source. Look in my picture above, the label CPU is part of the data object and is on the left, the other labels that are independent of a data object for a given source end up on the right no matter how I place them.

exelban, 2021-11-14:
Hi. Sorry but I don't understand and don't see the problem here. If you order the items manually, the macOS control the order of items. If it does not work, I can't help with that.

closed 2021-11-14 by exelban