exelban/stats #688: App crash on 11.6.1 - Illegal instruction: 4

roratonet, 2021-11-05:
The app is crashing on macOS 11.6.1 on MacBook Pro with the following details:

Model name: MacBook Pro
Model identification: MacBookPro11.1
Processor type: Dual-Core Intel Core i5
Processor speed: 2.8 GHz
Number of processors: 1
Total number of cores: 2
L2 cache (per core): 256 KB
L3 cache: 3 MB
Hyper-Threading Technology: Activated
Memory: 8 GB
System firmware version: 432.40.8.0.1
SMC version (system): 2.16f68
Serial number (system): Unavailable

[Stats_2021-11-05-191821_pedi Kopie.txt](https://github.com/exelban/stats/files/7488143/Stats_2021-11-05-191821_pedi.Kopie.txt)

exelban, 2021-11-05:
Hi. Thanks for reporting.
How it's possible that your mac does not have serial number?

exelban, 2021-11-05:
Could you please try this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/7488278/Stats.dmg.zip)

roratonet, 2021-11-05:
This is working fine.

Some old repaired MacBooks are loosing their S/N on the System. I don't know why 😞

closed 2021-11-05 by roratonet

exelban, 2021-11-05:
I was sure that every mac must have SN. But thanks for the report. This fix will be in the next release. Till then you can use this build.