exelban/stats #690: Sensor toggle button to enable/disable sensor on menu is inverted

jackblk, 2021-11-08:
**Describe the bug**

Sensor toggle button to enable/disable sensor is inverted

Sensor disabled:
[image]

Sensor enabled:
[image]

**Details:**
 - Device: Macbook Air M1 2020 16GB
 - macOS: 12.0.1
 - Application version: 2.6.18

exelban, 2021-11-08:
Hi. Thanks for the feedback.
I will check it.

I have noticed that you have empty widgets preview boxes. Do they disappear after the last update?

jackblk, 2021-11-08:
Wow what a fast reply!

About your question, I'm not sure because this is a fresh install of Stats via Homebrew.

jackblk, 2021-11-08:
@exelban I just updated Stats from 2.6.16 to 2.6.18 on my Mac 2017 (Big Sur 11.6), I can confirm that the preview is empty now after the update. Also, the Sensor toggle button is correct, even after update and I'm not sure why 😕

exelban, 2021-11-08:
Very strange.
I have 2 intel-based systems: 12.0 and 10.15. And everything works fine. I suppose that I know the reversed button logic reason. But have no idea why the widget preview is empty.

I will try to fix it.

jackblk, 2021-11-08:
I think the reversed button logic is not consistent, it will be fixed if I just quit & reopen it. But the empty preview is happening on both 11.6 (intel) & 12.0.1 (apple silicone) with Stats 2.7.18.

Btw the Power sensor is not available in Apple Silicone?

exelban, 2021-11-08:
On Apple Silicon temperature, current, and voltage sensors are available.

jackblk, 2021-11-08:
Oh I mean this part on Intel

[image]

I would love to have System Total (W) on Apple Silicon too. But thanks anyway :D

jackblk, 2021-11-11:
@exelban Works for me!

[image]

closed 2021-11-14 by exelban