exelban/stats #691: fan speed overlap

tom-NI, 2021-11-08:
I use the fan speed indicator that shows both fan speeds (v.2.6.18)

On my mac, macOS 11.6.1, the right side of the fan speed stats is cut off as the "box" the fan speeds are declared in does not appear to be large enough. No other stats appear to have overlap issues. This is happening with multiple other indicators to the right hand side of the fan speed indicator. 
Two rows, static width.
When i turn off static width, the size of the box is fine.

Increase the minimum size of the containing box for fan speeds for the above settings when the width is set to static, thanks

exelban, 2021-11-08:
Hi. Thanks for the feedback.
Are you using sensors or fans module?

tom-NI, 2021-11-09:
Fans module, apologies for the late reply

exelban, 2021-11-15:
Hi. Could I ask you to share the screenshot of the bug?

tom-NI, 2021-11-16:
[image]

closed 2021-11-28 by exelban