exelban/stats #692: Can't switch network card

johe123qwe, 2021-11-09:
**Describe the bug**
Can't switch network card, Only show one network card. The previous version can choose the network card.

**Details:**
 - Device: [Macbook Pro 2015]
 - macOS: 10.15.7
 - Application version: [2.6.18]

exelban, 2021-11-09:
Hi. Thanks for the feedback.
There were no changes related to this module for a while. Could you please install the previous version and check if it shows a few network interfaces?

johe123qwe, 2021-11-12:
Hi, Thank you for your reply.
I tested 2.6.17 and the network interface was fine, I could see multiple network interfaces. when I updated to 2.6.18 again, I still could not switch network interfaces.
https://take.ms/SRmeI

closed 2021-11-14 by exelban