exelban/stats #695: Add Celsius choice for sensors

movinimage, 2021-11-10:
I love Fahrenheit and all...but a lot of people are used to seeing CPU temps in Celsius. Would it be possible to add a choice to the sensors widget? Thanks for creating this awesome app.

exelban, 2021-11-11:
Hi. Thanks for the feedback. You can do it in the settings.

closed 2021-11-11 by exelban

movinimage, 2021-11-11:
Oh wow…I looked all around…Geez. Sorry to waste your time :)

nochkin, 2022-10-22:
Sorry for bringing it up. I must be going crazy, but where do I find it? Is it under "Module Settings"?
Right under "Update Interval" field I have "Save the fan speed" and other toggle options, but no temperature unit.

[image]