exelban/stats #698: [Feature Request] Show read write speeds for multiple disks.

LampPrinter, 2021-11-13:
I use stats to show read/write transfer speeds. But right now you can only select 1 disk to show stats (I have selected Internal). I always have an external drive connected, it would be nice if you can select multiple disks and it shows the combined read and write speed.
Also a threshold, maybe do not show speeds less than 1-2MBps would be great.

exelban, 2021-11-14:
Hi. Thanks for the feedback.
The first two features are already described in the issues. The threshold for speed? Why did it need? I don't see any reason why it must be implemented.

LampPrinter, 2021-11-14:
It is fine. The first 2 ones are the important ones ;)

exelban, 2021-11-14:
TLDR: first cannot be implemented, the second could be. But I need a design for it.

LampPrinter, 2021-11-14:
This is how iStats does it.

Per Disk

[image]

Combined

[image]

closed 2021-11-14 by exelban