exelban/stats #699: Fan control show "automatic" after reboot, should show "snowflake"

richardtop, 2021-11-15:
**Describe the bug**
When fans were set to max (*), the fan control will show "Automatic" after the reboot. The fans will still spin at max speed.

**Before the reboot:**

[image]

**After the reboot:**

[image]

**Solution:**
Clicking on the (*) symbol and back to "Automatic" actually sets the control back to automatic.

**Details:**
 - Device: Macbook Pro 16 2019
 - macOS: 12.0.1
 - Application version: 2.6.19

exelban, 2021-11-15:
Hi. Thanks for the feedback.
I will take a looks at this bug.

exelban, 2021-11-19:
Hi. It looks like some bug in the SMC. Because normally macOS will reset fans when reboot. It works fine in sleep mode. The reason why you see `Automatic` mode after reboot, it's because the fans report that they are in the auto mode. But for some reason, the fan's value was not reset before rebooting by macos.
I'm not sure I can do anything with it.

richardtop, 2021-11-19:
Ok, then feel free to close it.

exelban, 2021-11-19:
I will keep tracking this bug and hope it will be fixed in the near future. If not, I will try to do something with this.

closed 2021-11-19 by exelban