exelban/stats #700: many temperature options are not available on M1 Max

hongfeiyang, 2021-11-15:
**Describe the bug**
I could not find options like 'highest temperature' under the sensor tab. It was present on M1 MacBooks, but not on M1 Max.
and the 'PMU tcal' temperature is stuck at 52 degree Celsius and does not change regardless of the system load, although I don't know what that means, but I think it should vary just like the other sensors? :)

[image]
**Details:**
 - Device: MacBook Pro 16 (2021)
 - macOS: macOS 12.0.1
 - Application version: 2.6.19

exelban, 2021-11-15:
Hi. Thanks for the feedback.
I'm not sure I can do something with this. If you don't have sensors here, it means that API does not return them.
If you would like to help with it, give me know. I will need some additional information to figure out why CPU sensors are not available.

hongfeiyang, 2021-11-16:
> Hi. Thanks for the feedback.
> I'm not sure I can do something with this. If you don't have sensors here, it means that API does not return them.
> If you would like to help with it, give me know. I will need some additional information to figure out why CPU sensors are not available.

Thanks @exelban , you can give me the command to check temperatures in terminal and I could run some test for you. On my old M1 machine the app does give me the highest CPU temperature and GPU temperature, although I see `powermetrics` returns fields are the same for m1 and m1 max.
I wonder how other apps did it since m1 does not have a SMC anymore, I assume they are not approximating the CPU temperature by CPU voltage?

webjib, 2021-11-16:
Actually, M1 / M1 Pro / M1 Max do have SMC. You can get more than 200 temperatures sensors from SMC. However, we need to get the meaning for all SMC keys, and nothing is public. I guess other apps have reversed-engineered the keys.

I've written a small app that read all SMC keys, if you'd like to test:
[M1Tests.1.2.zip](https://www.jibapps.com/ressources/trashme3/files/M1Tests.1.2.zip)

exelban, 2021-11-16:
Could someone share the output of `/Applications/Stats.app/Contents/Resources/smc list`?

yitwail, 2021-11-16:
[smclist.txt](https://github.com/exelban/stats/files/7549377/smclist.txt)
Hope that works.

exelban, 2021-11-16:
Yeap, thanks. I will build a special version of Stats which will fetch sensors from the SMC. Will take a look if it works.

exelban, 2021-11-16:
Please try someone this build and give feedback if there are available sensors.
[Stats.dmg.zip](https://github.com/exelban/stats/files/7549758/Stats.dmg.zip)

hongfeiyang, 2021-11-16:
> Please try someone this build and give feedback if there are available sensors.
> [Stats.dmg.zip](https://github.com/exelban/stats/files/7549758/Stats.dmg.zip)

This is all I got for this build
[image]

hongfeiyang, 2021-11-16:
> Actually, M1 / M1 Pro / M1 Max do have SMC. You can get more than 200 temperatures sensors from SMC. However, we need to get the meaning for all SMC keys, and nothing is public. I guess other apps [...]

and here are the keys from your test app:

Model: MacBookPro18,2

---------------------------------------------------
FANS
---------------------------------------------------

Fan #0		0 rpm
Fan #1		0 rpm

---------------------------------------------------
SMC
---------------------------------------------------

Found 247 temperature keys in SMC

TAOL		24.37500 °C
TB0T		29.70000 °C
TB1T		29.20000 °C
TB2T		29.70000 °C
TC10		39.46253 °C
TC11		39.96501 °C
TC12		39.99852 °C
TC13		39.91477 °C
TC20		39.42904 °C
TC21		40.35026 °C
TC22		39.07730 °C
TC23		40.03201 °C
TC30		39.32854 °C
TC31		40.68525 °C
TC32		38.87630 °C
TC33		39.66353 °C
TC40		40.21626 °C
TC41		41.45574 °C
TC42		40.81924 °C
TC43		41.08725 °C
TC50		40.68526 °C
TC51		40.85275 °C
TC52		41.25475 °C
TC53		41.10400 °C
TCDX		41.89123 °C
TCHP		36.65767 °C
TCMb		51.62519 °C
TCMz		57.74500 °C
TD00		24.68914 °C
TD01		24.98886 °C
TD02		24.20498 °C
TD03		24.10117 °C
TD04		23.75178 °C
TD10		24.83275 °C
TD11		26.44447 °C
TD12		25.83618 °C
TD13		26.37903 °C
TD14		24.43458 °C
TD20		26.44313 °C
TD21		29.50269 °C
TD22		27.94513 °C
TD23		29.62503 °C
TD24		25.89995 °C
TDBP		28.68750 °C
TDCR		30.37500 °C
TDEC		29.93750 °C
TDEL		28.50000 °C
TDER		28.62500 °C
TDTP		31.81250 °C
TDVx		29.62503 °C
TDeL		28.25000 °C
TDeR		27.62500 °C
TH0a		32.44922 °C
TH0b		32.28906 °C
TH0x		32.44922 °C
TH1a		0.00000 °C
TH1b		0.00000 °C
TMVR		37.95045 °C
TPD0		38.10641 °C
TPD1		37.66241 °C
TPD2		38.01761 °C
TPD3		38.19521 °C
TPD4		38.01761 °C
TPD5		37.84001 °C
TPD6		37.92880 °C
TPD7		37.92880 °C
TPD8		38.01761 °C
TPD9		38.19521 °C
TPDX		38.19521 °C
TPDa		37.84001 °C
TPDb		37.66241 °C
TPDc		37.66241 °C
TPDd		37.84001 °C
TPDe		38.19521 °C
TPDf		37.75121 °C
TPDg		37.48480 °C
TPDh		37.48480 °C
TPDi		38.10641 °C
TPDj		37.75121 °C
TPMP		38.15767 °C
TPSP		38.51352 °C
TPVD		37.60986 °C
TRD0		39.43840 °C
TRD1		38.90561 °C
TRD2		39.17201 °C
TRD3		39.34961 °C
TRD4		39.52721 °C
TRD5		39.43840 °C
TRD6		39.26080 °C
TRD7		39.17201 °C
TRD8		39.26080 °C
TRD9		39.43840 °C
TRDX		39.61601 °C
TRDa		39.52721 °C
TRDb		39.17201 °C
TRDc		39.17201 °C
TRDd		39.17201 °C
TRDe		39.61601 °C
TRDf		39.08321 °C
TRDg		38.99440 °C
TRDh		38.81680 °C
TRDi		39.43840 °C
TRDj		39.17201 °C
TS0P		38.09460 °C
TSVR		37.11263 °C
TSWR		37.25226 °C
TSXR		36.68469 °C
TVA0		26.10623 °C
TVD0		51.62519 °C
TVMD		0.00000 °C
TVS0		32.65815 °C
TVS1		32.86613 °C
TVS2		32.98022 °C
TVSx		32.98022 °C
TVV0		41.49447 °C
TW0P		37.83334 °C
Ta00		0.00000 °C
Ta01		0.00000 °C
Ta02		0.00000 °C
Ta04		0.00000 °C
Ta05		8.48800 °C
Ta06		10.89600 °C
Ta08		0.00000 °C
Ta09		0.00000 °C
Ta0A		0.00000 °C
Ta0C		0.00000 °C
Ta0D		8.40100 °C
Ta0E		10.80700 °C
TaLP		37.12613 °C
TaLT		33.45985 °C
TaLW		33.04016 °C
TaRF		38.25226 °C
TaRT		30.59854 °C
TaRW		32.55475 °C
TaTP		38.14415 °C
Td00		38.40016 °C
Td01		38.40016 °C
Td02		39.35938 °C
Td04		38.59375 °C
Td05		38.59375 °C
Td06		40.09375 °C
Td08		38.10078 °C
Td09		38.10078 °C
Td0A		39.09375 °C
Td0C		37.77391 °C
Td0D		37.77391 °C
Td0E		38.82812 °C
Td0G		38.16609 °C
Td0H		38.16609 °C
Td0I		39.64062 °C
Td0K		36.68375 °C
Td0L		36.68375 °C
Td0M		37.50000 °C
Te00		38.26719 °C
Te01		51.62519 °C
Te02		57.74500 °C
Tg04		36.60760 °C
Tg05		45.80761 °C
Tg0C		35.57120 °C
Tg0D		44.77120 °C
Tg0K		37.25950 °C
Tg0L		46.45950 °C
Tg0S		36.40689 °C
Tg0T		45.60690 °C
Th00		37.74689 °C
Th01		45.07989 °C
Th02		47.14062 °C
Th04		38.10553 °C
Th05		38.10553 °C
Th06		38.35938 °C
Th08		38.10606 °C
Th09		38.10606 °C
Th0A		38.40625 °C
Th0C		37.23232 °C
Th0D		44.56532 °C
Th0E		46.50000 °C
Th0G		37.32132 °C
Th0H		37.32132 °C
Th0I		37.60938 °C
Tm00		36.39703 °C
Tm01		36.39703 °C
Tm02		37.39062 °C
Tm04		35.88438 °C
Tm05		35.88438 °C
Tm06		37.01562 °C
Tm08		37.72719 °C
Tm09		37.72719 °C
Tm0A		38.90625 °C
Tm0C		36.32563 °C
Tm0D		36.32563 °C
Tm0E		37.29688 °C
Tp00		36.59444 °C
Tp01		45.59044 °C
Tp02		51.77962 °C
Tp04		37.60000 °C
Tp05		46.57700 °C
Tp06		51.02600 °C
Tp08		39.30728 °C
Tp09		48.28528 °C
Tp0A		51.70425 °C
Tp0C		38.99471 °C
Tp0D		48.14471 °C
Tp0E		52.87312 °C
Tp0G		38.24660 °C
Tp0H		47.34860 °C
Tp0I		51.53587 °C
Tp0K		34.72464 °C
Tp0L		43.56564 °C
Tp0M		48.09087 °C
Tp0O		38.34121 °C
Tp0P		47.23621 °C
Tp0Q		51.84725 °C
Tp0S		37.75080 °C
Tp0T		46.81181 °C
Tp0U		50.44100 °C
Tp0W		39.43203 °C
Tp0X		48.36303 °C
Tp0Y		52.77913 °C
Tp0a		37.76506 °C
Tp0b		46.75407 °C
Tp0c		51.19513 °C
Ts00		37.94104 °C
Ts01		37.94104 °C
Ts02		39.14062 °C
Ts04		39.86074 °C
Ts05		39.86074 °C
Ts06		41.04688 °C
Ts0P		29.99609 °C
Ts1P		27.99609 °C
Tz11		0.00000 °C
Tz12		0.00000 °C
Tz13		0.00000 °C
Tz14		0.00000 °C
Tz15		0.00000 °C
Tz16		0.00000 °C
Tz17		0.00000 °C
Tz18		0.00000 °C
Tz1j		0.00000 °C

---------------------------------------------------
IOHID
---------------------------------------------------

gas gauge battery		29.70000 °C
NAND CH0 temp		32.00000 °C
PMU tcal		51.85001 °C
PMU tdev1		37.46397 °C
PMU tdev2		37.46397 °C
PMU tdev3		37.14865 °C
PMU tdev4		31.54381 °C
PMU tdev5		30.78104 °C
PMU tdev6		37.42793 °C
PMU tdev7		38.62163 °C
PMU tdev8		36.94145 °C
PMU tdie0		38.10641 °C
PMU tdie1		37.84001 °C
PMU tdie10		37.92880 °C
PMU tdie2		38.19521 °C
PMU tdie3		38.10641 °C
PMU tdie4		38.19521 °C
PMU tdie5		37.92880 °C
PMU tdie6		37.84001 °C
PMU tdie7		38.10641 °C
PMU tdie8		37.84001 °C
PMU tdie9		38.10641 °C
PMU TP0s		37.66241 °C
PMU TP1g		37.92880 °C
PMU TP1s		37.48480 °C
PMU TP2g		38.19521 °C
PMU TP2s		37.75121 °C
PMU TP3g		37.92880 °C

exelban, 2021-11-17:
So sorry, the wrong build. There is a good one.
[Stats.dmg.zip](https://github.com/exelban/stats/files/7552677/Stats.dmg.zip)

webjib, 2021-11-17:
@exelban what are we suppose to see? A new kind of window with SMC values?

exelban, 2021-11-17:
nope, just sensors

exelban, 2021-11-17:
I'm interested if there are any sensors in the Sensors module.

webjib, 2021-11-17:
I guess you've added SMC sensors only for M1 Max and M1 Pro? Because on a M1, there's nothing more than the IOHID sensors.

exelban, 2021-11-17:
Could you please share the screenshot of Sensors module popup window?

webjib, 2021-11-17:
Here we go (version 2.6.19, build 326)

[image]
[image]
[image]

exelban, 2021-11-17:
@webjib are you sure that it's a screenshot from the version that I was posted?
It's a little bit impossible that it has a `CPU performance core`.

@yitwail @hongfeiyang could you please try the build and was posted, and share the screenshot of the popup view of the Sensors module (not settings, but windows that appear when you click in the menu bar)?

exelban, 2021-11-17:
This build must be better.
[Stats.dmg.zip](https://github.com/exelban/stats/files/7557032/Stats.dmg.zip)

webjib, 2021-11-17:
With latest version 
(please note that I have a MacBook Air M1, and not a M1 Max / Pro)
[image]

casperes1996, 2021-11-17:
On an M1 Max (24 core GPU) with the latest build posted in this thread (16")
Looking a lot better! 
[image]

[image]

yitwail, 2021-11-18:
M1 MBA (plugged in
[image]
):

hongfeiyang, 2021-11-18:
> This build must be better.
> [Stats.dmg.zip](https://github.com/exelban/stats/files/7557032/Stats.dmg.zip)
[image]

hongfeiyang, 2021-11-18:
> So sorry, the wrong build. There is a good one.
> [Stats.dmg.zip](https://github.com/exelban/stats/files/7552677/Stats.dmg.zip)
[image]

webjib, 2021-11-18:
@exelban I've tried to find which SMC keys to observe for M1 / M1 Pro / M1 Max, but I can't find any information. Some other apps like TG Pro and Macs Fan Control claim to support new M1 chips, probably thanks to SMC as the IOKit is pretty useless on M1 Pro/Max. I guess they will not share their findings, but maybe you can try to contact them?

casperes1996, 2021-11-18:
> @exelban I've tried to find which SMC keys to observe for M1 / M1 Pro / M1 Max, but I can't find any information. Some other apps like TG Pro and Macs Fan Control claim to support new M1 chips, [...]

Given the projects are paid (somewhat at least for Macs Fan Control) and closed source I doubt that'll give anything. Maybe Macs Fan Control as its free version is pretty capable and the dev seems fairly chill but I wouldn't hold my breath for TG Pro. 
Hopefully Apple updates IOKit, but reverse engineering the SMC keys should be possible too though a bit of a hassle and time sink potentially

exelban, 2021-11-18:
Hi. Thanks for the proposals. I'm pretty sure that no one will give a list of SMC keys. Even now, there is no good list for intel based. I have somewhere the app that fetches the SMC keys names from the system. Will try to find it, and share it. It could help to create a map key <-> sensor name.

webjib, 2021-11-18:
@casperes1996 you're probably right about "competitors" :-/ About Apple, I'm not sure they will update IOKit to add same keys we can find for the M1 SoC.

@exelban I've tried to extract the SMC keys names from libSMC.dylib, but without luck, there's only old SMC keys for Intel ; please try and see if you can extract something.

exelban, 2021-11-18:
I don't have any apple silicon, only intel based(

webjib, 2021-11-18:
Please share your app that fetch SMC names, I'll be glad to share results.

exelban, 2021-11-18:
Yeap, I will post it later today.

casperes1996, 2021-11-18:
I have an M1 Max machine - with instructions I'll help where I can

webjib, 2021-11-19:
If someone with a M1 Max/Pro would like to test this new version of the test app I've coded, it will dump all SMC keys from the system, with theirs names (first text box). On my MBA M1, this does not help, but who knows, it's maybe better with M1 Pro/Max :-)
[M1Tests.1.3.zip](https://www.jibapps.com/ressources/trashme3/files/M1Tests.1.3.zip)

casperes1996, 2021-11-19:
> If someone with a M1 Max/Pro would like to test this new version of the test app I've coded, it will dump all SMC keys from the system, with theirs names (first text box). On my MBA M1, this does [...]

Hi! The following is from an M1 Max with 24 GPU cores (16")

`TA0p => Ambient raw temp (DegC) (TA0p)
TA0P => Ambient cooked temp (DegC) (TA0P)
Ta0P => air flow temp (DegC) (Ta0P)
TA0V => Ambient virtual temp (DegC) (TA0V)
TA1P => Ambient cooked temp 1 (DegC) (TA1P)
TA1p => Ambient 2 raw temp (DegC) (TA1p)
TA2P => Ambient cooked temp 2 (DegC) (TA2P)
TA2p => Ambient 3 raw temp (DegC) (TA2p)
TaSP => DCIn Air flow Adjusted Temperature (DegC) (TaSP)
Tb0P => Backlight Controller proximity cooked temp (DegC) (Tb0P)
Tb0p => Backlight Controller proximity raw temp (DegC) (Tb0p)
TB0T => Battery temp (SBS! force bit 0) (DegC) (TB0T)
TB1T => Battery TS1 Temp (DegC) (TB1T)
TB2T => Battery TS2 Temp (DegC) (TB2T)
TB3T => Battery Ambient Temp (TB3T)
TBXT => Battery temp (Same as TB0T) (DegC) (TBXT)
TC0c => CPU Core 0 absolute raw temp (PECI) (DegC) (TC0c)
TC0C => CPU Core 0 temp (PECI) (DegC) (TC0C)
TC0D => CPU die temp (DegC) (TC0D)
TC0d => CPU die raw temp (DegC) (TC0d)
TC0E => CPU die Virtual temp (DegC) (TC0E)
TC0F => CPU PECI die temp filtered then adjusted (DegC) (TC0F)
TC0J => CPU PECI Die temp max error used in TC0F=TC0E+ (DegC) (TC0J)
TC0L => CPU PECI Die dynamic filter coeff (Seconds) (TC0L)
TC0P => CPU proximity temp (DegC) (TC0P)
TC0p => CPU Proximity raw temp (DegC) (TC0p)
TC0T => CPU PECI Die temp Trend (DegC) (TC0T)
TC1c => CPU Core 1 absolute raw temp (PECI) (DegC) (TC1c)
TC1C => CPU Core 0 temp (PECI) (DegC) (TC1C)
TC1P => CPU VR Proximity cooked temp (DegC) (TC1P)
TC2c => CPU Core 2 absolute raw temp (PECI) (DegC) (TC2c)
TC2C => CPU Core 1 temp (PECI) (DegC) (TC2C)
TC3C => CPU Core 2 temp (PECI) (DegC) (TC3C)
TC3c => CPU Core 3 absolute raw temp (PECI) (DegC) (TC3c)
TC4C => CPU Core 3 temp (PECI) (DegC) (TC4C)
TCFC => CPU PECI die temp filter coeff (TCFC)
TCGC => CPU Gfx Core temp (PECI) (DegC) (TCGC)
TCGc => CPU Gfx absolute raw temp (PECI) (DegC) (TCGc)
TCHP => Charger proximity temp  (DegC) (TCHP)
TCMc => CPU eDRAM Hot Indicator(PECI) (DegC) (TCMc)
TCMX => Max PECI reported temp  (DegC) (TCMX)
TCSA => CPU System Agent Core temp (PECI) (DegC) (TCSA)
TCSC => CPU System Agent Core absolute cooked temp (PECI) (DegC) (TCSC)
TCSc => CPU System Agent Core absolute raw temp (PECI) (DegC) (TCSc)
TCTD => CPU PECI die temp Trend (DegC) (TCTD)
TCXC => CPU Core PECI temp (DegC) (SIS! force bit 7) (TCXC)
TCXc => CPU Max Package Core absolute raw temp (PECI) (DegC) (TCXc)
TCXr => CPU Max Package Core relative raw temp (PECI) (DegC) (TCXr)
TCXR => CPU Max Package Core relative cooked temp (PECI) (DegC) (TCXR)
Te0T => Te0T
Te0t => TBT Thermal Diode raw temp (DegC) (Te0t)
TG0d => GPU Die raw temp (DegC) (TG0d)
TG0D => GPU Die cooked temp (DegC) (TG0D)
TG0E => GPU Die filtered temp (DegC) (TG0E)
TG0F => GPU Die filtered and adjusted temp for fan/power control (DegC) (TG0F)
TG0H => Heat Pipe Temp(TG0H)
TG0J => GPU Die temp max error used in TG0F=TG0E+ (DegC) (TG0J)
TG0L => GPU Die dynamic filter coeff (Seconds) (TG0L)
TG0M => GFX memory temp (DegC) (TG0M)
TG0P => GPU Proximity 0 (internal EMC1414) cooked temp (DegC) (TG0P)
TG0p => GPU Proximity 0 (internal EMC1414) raw temp (DegC) (TG0p)
TG0T => GPU Die temp Trend (DegC) (TG0T)
TG1d => GPU digital die temp Raw (without offset) (DegC) (TG1d)
TG1D => GPU B Die cooked temp (DegC) (TG1D)
TG1F => GPU digital die temp filtered (DegC) (TG1F)
TG1H => Heat Pipe Temp2(TG1H)
[... 383 lines cut ...]
PMU tdev2		34,63139 °C
PMU tdev3		34,49271 °C
PMU tdev4		30,05840 °C
PMU tdev5		30,25912 °C
PMU tdev6		33,64235 °C
PMU tdev7		34,40512 °C
PMU tdev8		34,52556 °C
PMU tdie0		35,35361 °C
PMU tdie1		35,53120 °C
PMU tdie10		35,26480 °C
PMU tdie2		35,35361 °C
PMU tdie3		35,53120 °C
PMU tdie4		35,79761 °C
PMU tdie5		35,53120 °C
PMU tdie6		35,26480 °C
PMU tdie7		35,44241 °C
PMU tdie8		35,70880 °C
PMU tdie9		35,79761 °C
PMU TP0s		35,17601 °C
PMU TP1g		35,53120 °C
PMU TP1s		35,17601 °C
PMU TP2g		35,70880 °C
PMU TP2s		35,26480 °C
PMU TP3g		35,53120 °C
`

casperes1996, 2021-11-19:
The following may be more readable for the names - same content as above, just laid out nicer :)

TA0p => Ambient raw temp (DegC) (TA0p)
TA0P => Ambient cooked temp (DegC) (TA0P)
Ta0P => air flow temp (DegC) (Ta0P)
TA0V => Ambient virtual temp (DegC) (TA0V)
TA1P => Ambient cooked temp 1 (DegC) (TA1P)
TA1p => Ambient 2 raw temp (DegC) (TA1p)
TA2P => Ambient cooked temp 2 (DegC) (TA2P)
TA2p => Ambient 3 raw temp (DegC) (TA2p)
TaSP => DCIn Air flow Adjusted Temperature (DegC) (TaSP)
Tb0P => Backlight Controller proximity cooked temp (DegC) (Tb0P)
Tb0p => Backlight Controller proximity raw temp (DegC) (Tb0p)
TB0T => Battery temp (SBS! force bit 0) (DegC) (TB0T)
TB1T => Battery TS1 Temp (DegC) (TB1T)
TB2T => Battery TS2 Temp (DegC) (TB2T)
TB3T => Battery Ambient Temp (TB3T)
TBXT => Battery temp (Same as TB0T) (DegC) (TBXT)
TC0c => CPU Core 0 absolute raw temp (PECI) (DegC) (TC0c)
TC0C => CPU Core 0 temp (PECI) (DegC) (TC0C)
TC0D => CPU die temp (DegC) (TC0D)
TC0d => CPU die raw temp (DegC) (TC0d)
TC0E => CPU die Virtual temp (DegC) (TC0E)
TC0F => CPU PECI die temp filtered then adjusted (DegC) (TC0F)
TC0J => CPU PECI Die temp max error used in TC0F=TC0E+ (DegC) (TC0J)
TC0L => CPU PECI Die dynamic filter coeff (Seconds) (TC0L)
TC0P => CPU proximity temp (DegC) (TC0P)
TC0p => CPU Proximity raw temp (DegC) (TC0p)
TC0T => CPU PECI Die temp Trend (DegC) (TC0T)
TC1c => CPU Core 1 absolute raw temp (PECI) (DegC) (TC1c)
TC1C => CPU Core 0 temp (PECI) (DegC) (TC1C)
TC1P => CPU VR Proximity cooked temp (DegC) (TC1P)
TC2c => CPU Core 2 absolute raw temp (PECI) (DegC) (TC2c)
TC2C => CPU Core 1 temp (PECI) (DegC) (TC2C)
TC3C => CPU Core 2 temp (PECI) (DegC) (TC3C)
TC3c => CPU Core 3 absolute raw temp (PECI) (DegC) (TC3c)
TC4C => CPU Core 3 temp (PECI) (DegC) (TC4C)
TCFC => CPU PECI die temp filter coeff (TCFC)
TCGC => CPU Gfx Core temp (PECI) (DegC) (TCGC)
TCGc => CPU Gfx absolute raw temp (PECI) (DegC) (TCGc)
TCHP => Charger proximity temp  (DegC) (TCHP)
TCMc => CPU eDRAM Hot Indicator(PECI) (DegC) (TCMc)
TCMX => Max PECI reported temp  (DegC) (TCMX)
TCSA => CPU System Agent Core temp (PECI) (DegC) (TCSA)
TCSC => CPU System Agent Core absolute cooked temp (PECI) (DegC) (TCSC)
TCSc => CPU System Agent Core absolute raw temp (PECI) (DegC) (TCSc)
TCTD => CPU PECI die temp Trend (DegC) (TCTD)
TCXC => CPU Core PECI temp (DegC) (SIS! force bit 7) (TCXC)
TCXc => CPU Max Package Core absolute raw temp (PECI) (DegC) (TCXc)
TCXr => CPU Max Package Core relative raw temp (PECI) (DegC) (TCXr)
TCXR => CPU Max Package Core relative cooked temp (PECI) (DegC) (TCXR)
Te0T => Te0T
Te0t => TBT Thermal Diode raw temp (DegC) (Te0t)
TG0d => GPU Die raw temp (DegC) (TG0d)
TG0D => GPU Die cooked temp (DegC) (TG0D)
TG0E => GPU Die filtered temp (DegC) (TG0E)
TG0F => GPU Die filtered and adjusted temp for fan/power control (DegC) (TG0F)
TG0H => Heat Pipe Temp(TG0H)
TG0J => GPU Die temp max error used in TG0F=TG0E+ (DegC) (TG0J)
TG0L => GPU Die dynamic filter coeff (Seconds) (TG0L)
TG0M => GFX memory temp (DegC) (TG0M)
TG0P => GPU Proximity 0 (internal EMC1414) cooked temp (DegC) (TG0P)
TG0p => GPU Proximity 0 (internal EMC1414) raw temp (DegC) (TG0p)
TG0T => GPU Die temp Trend (DegC) (TG0T)
TG1d => GPU digital die temp Raw (without offset) (DegC) (TG1d)
TG1D => GPU B Die cooked temp (DegC) (TG1D)
TG1F => GPU digital die temp filtered (DegC) (TG1F)
TG1H => Heat Pipe Temp2(TG1H)
TG1p => GPU GDDR5 Proximity 1 raw temp (DegC) (TG1p)
TG1P => GPU GDDR5 Proximity 1 cooked temp (DegC) (TG1P)
[... 92 lines cut ...]
TP0P => PCH proximity temp (DegC) (TP0P)
Tp0T => PSU Secondary H/S Diode cooked temp (DegC) (Tp0T)
Tp1P => Secondary power supply proximity (Tp1P)
Tp2F => Power Supply T2 Secondary Heatsink filtered temp (DegC) (Tp2F)
Tp2H => Power Supply T2 Secondary Heatsink cooked temp (DegC) (Tp2H)
Tp2h => Power Supply T2 Secondary Heatsink raw temp (DegC) (Tp2h)
TPCd => PCH digital die raw temp (DegC) (TPCd)
TPCD => PCH Die temp (DegC) (TPCD)
TpFP => Filtered Power supply 1 proximity (TpFP)
TS0D => Speaker Remote Temp (DegC) (TS0D)
Ts0P => Trackpad Temp (DegC) (Ts0P)
TS0p => Skin 0 raw temp (DegC) (TS0p)
TS0P => Speaker Proximity Temp (DegC) (TS0P)
Ts0S => Synthetic bottom skin (DegC) (Ts0S)
TS0V => Skin 0 virtual temp (DegC) (TS0V)
TS1D => Speaker Remote 2 Temp (DegC) (TS1D)
TS1P => Speaker Proximity 2 Temp (DegC) (TS1P)
Ts1P => Actuator Temp (DegC) (Ts1P)
Ts1S => Synthetic top skin (DegC) (Ts1S)
TS2P => S2 camera proximity temp (DegC) (TS2P)
TTF0 => Fan Target Temp(TTF0)
TV0P => Vent proximity temp (TV0P)
TVFP => Filtered Vent proximity (TVFP)
TW0p => Wifi proximity raw temp (DegC) (TW0p)
TW0P => Wireless proximity (DegC) (TW0P)

webjib, 2021-11-19:
Thanks! So, nothing more than @exelban already found :-/
I think that Apple stopped to maintain the human readable list of SMC Keys.

exelban, 2021-11-19:
Hi. Thanks for the feedback. But please do not spam thread. @casperes1996 if you need to share a lot of text, please use the files.

casperes1996, 2021-11-19:
Sorry - will make sure to keep that in mind going forward :)

exelban, 2021-11-19:
@webjib thanks. I suppose your app read the names from the `libSMC.dylib`?

webjib, 2021-11-19:
@exelban exactly!

exelban, 2021-11-19:
I have the same way. So it looks like there is no option to fetch the names of the sensors from macOS.

JoshuaChen, 2021-11-22:
`TA0P => Ambient cooked temp (DegC) (TA0P)
Ta0P => air flow temp (DegC) (Ta0P)
TA0p => Ambient raw temp (DegC) (TA0p)
TA0V => Ambient virtual temp (DegC) (TA0V)
TA1p => Ambient 2 raw temp (DegC) (TA1p)
TA1P => Ambient cooked temp 1 (DegC) (TA1P)
TA2P => Ambient cooked temp 2 (DegC) (TA2P)
TA2p => Ambient 3 raw temp (DegC) (TA2p)
TaSP => DCIn Air flow Adjusted Temperature (DegC) (TaSP)
Tb0P => Backlight Controller proximity cooked temp (DegC) (Tb0P)
Tb0p => Backlight Controller proximity raw temp (DegC) (Tb0p)
TB0T => Battery temp (SBS! force bit 0) (DegC) (TB0T)
TB1T => Battery TS1 Temp (DegC) (TB1T)
TB2T => Battery TS2 Temp (DegC) (TB2T)
TB3T => Battery Ambient Temp (TB3T)
TBXT => Battery temp (Same as TB0T) (DegC) (TBXT)
TC0c => CPU Core 0 absolute raw temp (PECI) (DegC) (TC0c)
TC0C => CPU Core 0 temp (PECI) (DegC) (TC0C)
TC0d => CPU die raw temp (DegC) (TC0d)
TC0D => CPU die temp (DegC) (TC0D)
TC0E => CPU die Virtual temp (DegC) (TC0E)
TC0F => CPU PECI die temp filtered then adjusted (DegC) (TC0F)
TC0J => CPU PECI Die temp max error used in TC0F=TC0E+ (DegC) (TC0J)
TC0L => CPU PECI Die dynamic filter coeff (Seconds) (TC0L)
TC0p => CPU Proximity raw temp (DegC) (TC0p)
TC0P => CPU proximity temp (DegC) (TC0P)
TC0T => CPU PECI Die temp Trend (DegC) (TC0T)
TC1c => CPU Core 1 absolute raw temp (PECI) (DegC) (TC1c)
TC1C => CPU Core 0 temp (PECI) (DegC) (TC1C)
TC1P => CPU VR Proximity cooked temp (DegC) (TC1P)
TC2C => CPU Core 1 temp (PECI) (DegC) (TC2C)
TC2c => CPU Core 2 absolute raw temp (PECI) (DegC) (TC2c)
TC3c => CPU Core 3 absolute raw temp (PECI) (DegC) (TC3c)
TC3C => CPU Core 2 temp (PECI) (DegC) (TC3C)
TC4C => CPU Core 3 temp (PECI) (DegC) (TC4C)
TCFC => CPU PECI die temp filter coeff (TCFC)
TCGc => CPU Gfx absolute raw temp (PECI) (DegC) (TCGc)
TCGC => CPU Gfx Core temp (PECI) (DegC) (TCGC)
TCHP => Charger proximity temp  (DegC) (TCHP)
TCMc => CPU eDRAM Hot Indicator(PECI) (DegC) (TCMc)
TCMX => Max PECI reported temp  (DegC) (TCMX)
TCSA => CPU System Agent Core temp (PECI) (DegC) (TCSA)
TCSC => CPU System Agent Core absolute cooked temp (PECI) (DegC) (TCSC)
TCSc => CPU System Agent Core absolute raw temp (PECI) (DegC) (TCSc)
TCTD => CPU PECI die temp Trend (DegC) (TCTD)
TCXc => CPU Max Package Core absolute raw temp (PECI) (DegC) (TCXc)
TCXC => CPU Core PECI temp (DegC) (SIS! force bit 7) (TCXC)
TCXr => CPU Max Package Core relative raw temp (PECI) (DegC) (TCXr)
TCXR => CPU Max Package Core relative cooked temp (PECI) (DegC) (TCXR)
Te0t => TBT Thermal Diode raw temp (DegC) (Te0t)
Te0T => Te0T
TG0d => GPU Die raw temp (DegC) (TG0d)
TG0D => GPU Die cooked temp (DegC) (TG0D)
TG0E => GPU Die filtered temp (DegC) (TG0E)
TG0F => GPU Die filtered and adjusted temp for fan/power control (DegC) (TG0F)
TG0H => Heat Pipe Temp(TG0H)
TG0J => GPU Die temp max error used in TG0F=TG0E+ (DegC) (TG0J)
TG0L => GPU Die dynamic filter coeff (Seconds) (TG0L)
TG0M => GFX memory temp (DegC) (TG0M)
TG0p => GPU Proximity 0 (internal EMC1414) raw temp (DegC) (TG0p)
TG0P => GPU Proximity 0 (internal EMC1414) cooked temp (DegC) (TG0P)
TG0T => GPU Die temp Trend (DegC) (TG0T)
TG1D => GPU B Die cooked temp (DegC) (TG1D)
TG1d => GPU digital die temp Raw (without offset) (DegC) (TG1d)
TG1F => GPU digital die temp filtered (DegC) (TG1F)
TG1H => Heat Pipe Temp2(TG1H)
TG1p => GPU GDDR5 Proximity 1 raw temp (DegC) (TG1p)
TG1P => GPU GDDR5 Proximity 1 cooked temp (DegC) (TG1P)
TG2P => GPU VR Proximity 2 cooked temp (DegC) (TG2P)
TG2p => GPU VR Proximity 2 raw temp (DegC) (TG2p)
[... 91 lines cut ...]
Tp0T => PSU Secondary H/S Diode cooked temp (DegC) (Tp0T)
Tp1P => Secondary power supply proximity (Tp1P)
Tp2F => Power Supply T2 Secondary Heatsink filtered temp (DegC) (Tp2F)
Tp2h => Power Supply T2 Secondary Heatsink raw temp (DegC) (Tp2h)
Tp2H => Power Supply T2 Secondary Heatsink cooked temp (DegC) (Tp2H)
TPCD => PCH Die temp (DegC) (TPCD)
TPCd => PCH digital die raw temp (DegC) (TPCd)
TpFP => Filtered Power supply 1 proximity (TpFP)
TS0D => Speaker Remote Temp (DegC) (TS0D)
Ts0P => Trackpad Temp (DegC) (Ts0P)
TS0P => Speaker Proximity Temp (DegC) (TS0P)
TS0p => Skin 0 raw temp (DegC) (TS0p)
Ts0S => Synthetic bottom skin (DegC) (Ts0S)
TS0V => Skin 0 virtual temp (DegC) (TS0V)
TS1D => Speaker Remote 2 Temp (DegC) (TS1D)
Ts1P => Actuator Temp (DegC) (Ts1P)
TS1P => Speaker Proximity 2 Temp (DegC) (TS1P)
Ts1S => Synthetic top skin (DegC) (Ts1S)
TS2P => S2 camera proximity temp (DegC) (TS2P)
TTF0 => Fan Target Temp(TTF0)
TV0P => Vent proximity temp (TV0P)
TVFP => Filtered Vent proximity (TVFP)
TW0P => Wireless proximity (DegC) (TW0P)
TW0p => Wifi proximity raw temp (DegC) (TW0p)
`

exelban, 2021-11-22:
thanks. But don't post the data that has already been presented here.

exelban, 2021-11-25:
@webjib hi, could you please share the output of `/Applications/Stats.app/Contents/Resources/smc list >> sensors.txt`?
And please, post your MacBook configuration, number of CPU cores, GPU, etc.

webjib, 2021-11-26:
I only have a MacBook Air M1, for which Stats already displays information about temperature. Do you really want the SMC list?

exelban, 2021-11-26:
@webjib yes, please share. I want to collect all keys which are available on apple silicon. And will try to make a dictionary.

exelban, 2021-11-26:
Could someone with apple m1 pro and max share the result of /Applications/Stats.app/Contents/Resources/smc list >> sensors.txt?
And please, post your CPU configuration, number of CPU cores, GPU, etc.

webjib, 2021-11-26:
**MacBook Air M1 (MacBookAir10,1)**
CPU: Apple M1 - 8 Core
GPU: Apple M1 - 7 core

[sensors.txt](https://github.com/exelban/stats/files/7608087/sensors.txt)

casperes1996, 2021-11-26:
**MacBook Pro M1 Max 16" (MacBook Pro18,2)**
CPU: Apple M1 Max - 10 Core (2 efficiency, 8 performance)
GPU Apple M1 Max - 24 Co
[sensors.txt](https://github.com/exelban/stats/files/7608188/sensors.txt)
re

exelban, 2021-11-26:
@webjib @casperes1996 thanks.

exelban, 2021-11-26:
@yitwail Could I ask what CPU do you have?

exelban, 2021-11-26:
Maybe someone has an M1 pro, to have a full list?

Now I see that pro/max has totally different sensors than m1.
And looks like max/pro will need a separate module for sensors. There is no one solution for m1/pro/max. So please, stop posting the information about m1, because this thread is only for m1 pro/max. The difference between m1 and m1 pro/max is the same as between m1 and Intel based. Based on data from the SMC (it's not a really SMC, but it looks like apple pretend that new MacBooks has it, or maybe they have it).

kennymc-c, 2021-11-26:
@exelban 

MacBook Pro M1 Pro 10 Core / 16 Core GPU

[sensors.txt](https://github.com/exelban/stats/files/7610260/sensors.txt)

jcastro, 2021-11-26:
I have a M1 Pro 8c 14GPU if that helps?

yitwail, 2021-11-27:
> Maybe someone has an M1 pro, to have a full list?
>
> Now I see that pro/max has totally different sensors than m1. And looks like max/pro will need a separate module for sensors. There is no one [...]

I missed this comment, but FYI I created a new issue originally for m1, and you closed it as a duplicate, here: [https://github.com/exelban/stats/issues/702#issue-1055135483](url)

exelban, 2021-11-27:
@yitwail yeap, I was thinking that it's related. But figured out that no. But still I cant help with m1. If it shows the wrong number of sensors it only means that system reports that sensors.

holdit, 2021-11-29:
MBP 16" 2021
M1 Max 10 CPU Cores
32 GPU Cores

[sensors.txt](https://github.com/exelban/stats/files/7614694/sensors.txt)

ugurozturk, 2021-11-29:
Apple M1 Pro 14"
8 cpu
14 gpu
[sensors.txt](https://github.com/exelban/stats/files/7621657/sensors.txt)

exelban, 2021-11-30:
Hi.
I've taken a look at the SMC keys shared here. And, unfortunately, don't identify any keys. I was trying to find some keys (temperature only) for CPU or GPU cores by the correlation between the number of cores and the number of keys with the same prefix. And the number of these keys must be different for each SoC.

For example: 
- 8 CPU: 6 power and 2 efficient cores
- 10 CPU: 8 power and 2 efficient cores
In this case, I was looking for the keys which will have the same group, but different numbers in the different configurations of the CPU. But don't find any consistency. The same trick I was trying to make with the parts which are different for each SoC: CPU cores, GPU cores, and neural cores.

@webjib proposed that `Tp0T` and `Tp01` it's efficient cores. But I do not agree with this. Because for each CPU there is 30 elements with `TpXX` prefix: `Tp00` - `Tp0c`. And I have no idea which of these elements equals the specific core.

There is a list of temperature keys that I separate by prefixes and count. Maybe someone will find some correlations between these sensors and the SoC parts.

[max-10:24.txt](https://github.com/exelban/stats/files/7628536/max-10.24.txt)
[pro-10:16.txt](https://github.com/exelban/stats/files/7628538/pro-10.16.txt)
[pro-8:14.txt](https://github.com/exelban/stats/files/7628539/pro-8.14.txt)

webjib, 2021-12-02:
So I guess we'll need to find more information about the appropriate keys. The only thing you can notice is that temperatures are coherent each 4 key. For exemple: Tp01, Tp05, Tp09, Tp0D... are coherent. The same for Tp00, Tp04, Tp08...

It seems that we have 10 groups of 3 keys (at least for CPU):
[Tp00, Tp01, Tp02]
[Tp04, Tp05, Tp06]
[Tp08, Tp09, Tp0A]
[Tp0C, Tp0D, Tp0E]
...
[Tp0a, Tp0b, Tp0c]

This is only an assumption, but the first value of each group could be the min value, the second could be the current value and the third could be the max value?

exelban, 2021-12-02:
Yeap, it make sense. First - minimum, second - current, and third - maximum.
Maybe you have an idea of how power and efficient cores could be found in these pairs?

[11 more replies]

closed 2022-01-16 by exelban