exelban/stats #705: Monospaced Fonts

mYoda, 2021-11-18:
Hello! I would like to suggest using monospaced fonts and adding label width minWidth. This will reduce "jumping" on the top bar

exelban, 2021-11-18:
Hi. Thanks for the feedback.
I was using a monospaced font. But on some macs it produces a crash.

mYoda, 2021-11-18:
I would like to investigate issue if it’s possible)
Probably this case we have to fix crash.

Best regards,
Anton

exelban, 2021-11-18:
[This](https://github.com/exelban/stats/commit/7ce7e38d36c0e31ee0c1007e1ad6c2a406345a99) commit. There were more issues related to this. But only this solution fixes all crashes. Not only on m1.

mYoda, 2021-11-18:
Yeah, thanks. Sometimes systemfont() might cause a crash. Would you like to add Fonts in a Bundle? 
Also I would like to suggest swiftgen to work with Fonts and to generate related Swift code…
I have good templates for Fonts, Strings(Localizable), Colors.

Best regards and stay healthy,
Anton

exelban, 2021-11-18:
I would like to have a monospaced font in the widgets for sensors or speed. Because it's easy to size the container. But when only one widget is monospaced it looks a little strange. So I think there is only one option, to use the monospaced font in all widgets. I'm not sure it will look nice (I don't like how it looks with a system monospaced font). But if you could make some version of Stats with a monospaced font, you can make a PR with it. And then will decide if it's will be added to the app.

Also I was thinking about font style toggle. But it will remove the big advantage of the monospaced font.

exelban, 2021-11-28:
Hi. Give me please to know if you will propose something.

153957, 2022-01-12:
Maybe you can enable 'tabular' figures for the non-monospaced font to still get the same font and styling, but with aligned numbers?

exelban, 2022-01-12:
What do you mean by "tabular"?

153957, 2022-01-13:
See this https://www.fonts.com/content/learning/fontology/level-3/numbers/proportional-vs-tabular-figures

Some fonts have tabular variants for digits to allow them to be aligned.

exelban, 2022-01-13:
Ok, but does macOS system font has this option?

153957, 2022-01-13:
Maybe this is the font you will need to use: [monospacedDigitSystemFont](https://developer.apple.com/documentation/appkit/nsfont/1528205-monospaceddigitsystemfont)

exelban, 2022-01-13:
I will not use the monospaced font, it causes crashes on some macs. Please read the answers above.

153957, 2022-01-13:
This font only uses monospaced digits, all other letter will be proportional, maybe that means it wont be an issue, it might be nice to try..

exelban, 2022-01-13:
Yeap, but unfortunately I cannot reproduce the crash with the font. And I will not try if it could cause a crashes in some users. I really like monospaced fonts over not-monospaced for some cases. But I prefer to have a working app than crashing on start.

exelban, 2022-02-07:
Hi. Please ping me if you would propose something. I will close this issue for now due to inactivity.

closed 2022-02-07 by exelban