exelban/stats #707: network speed without pictogram takes unnecessary space

chen3326, 2021-11-19:
The network speed widget still takes the same amount of space when the pictogram option is set to none. This makes half of the space taken to be wasted. Could you consider changing that behavior?

hongfeiyang, 2021-11-19:
Yea I also found many of the menu bar icons taking up too much space, I really would like to have the icons resize themselves to the minimum space they could take. I have a notched MacBook Pro and the status bar space is getting more an more precious now :) @exelban

exelban, 2021-11-19:
Hi. Thanks for the feedback.
If you have disabled the pictogram option, it will remove the space which needed to show the pictogram. And leave the minimum space which needed to show the value.

closed 2021-11-19 by exelban