exelban/stats #712: Interface and Network unavailable

IcedDragon, 2021-11-22:
Hi!

I noticed that on my network menu on system tray, the Interface and Network are not displayed. Also the memory and graphics info is not displayed.
I'm sending the printscreens. 
My macbook pro 15'' is Mid 2014.

Beste regards!
Luis Marques

[image]

[image]

exelban, 2021-11-24:
Hi. Thanks for the feedback.
Would you like to help to fix this?

IcedDragon, 2021-11-24:
Hi

Yes, of course. 
What do I need to do?

Luis Marques

exelban, 2021-11-24:
Please run these command and put the output in some file, and share it.

For RAM information `/usr/sbin/system_profiler SPMemoryDataType -json`.
For GPU information `/usr/sbin/system_profiler SPDisplaysDataType -json `.

IcedDragon, 2021-11-24:
> <?xml version="1.0" encoding="UTF-8"?>
> <!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
> <plist version="1.0">
> <array>
> 	<dict>
> 		<key>_SPCommandLineArguments</key>
> 		<array>
> 			<string>/usr/sbin/system_profiler</string>
> 			<string>-nospawn</string>
> 			<string>-xml</string>
> 			<string>SPMemoryDataType</string>
> 			<string>-detailLevel</string>
> 			<string>full</string>
> 		</array>
> 		<key>_SPCompletionInterval</key>
> 		<real>0.045378923416137695</real>
> 		<key>_SPResponseTime</key>
> 		<real>0.12835001945495605</real>
> 		<key>_dataType</key>
> 		<string>SPMemoryDataType</string>
> 		<key>_detailLevel</key>
> 		<integer>-1</integer>
> 		<key>_items</key>
> 		<array>
> 			<dict>
> [... 562 lines cut ...]
> 			<string>536</string>
> 		</dict>
> 	</dict>
> </array>
> </plist>

HelloI can only output in XMLI save those on to a TXT file.Regards,Luis Marques

exelban, 2021-11-24:
Thanks. You cannot export the output as `json` because the `system_profiler` does not support `-json` option?

IcedDragon, 2021-11-24:
This is what it outputs on terminal when I choose JSON-------/usr/sbin/system_profiler SPDisplaysDataType -jsonUsage: system_profiler [-listDataTypes]       system_profiler [-xml] [-timeout n] [-detailLevel n]       system_profiler [-xml] [-timeout n] [dataType1 ... dataTypeN]  -detailLevel n    specifies the level of detail for the report                      mini = short report (contains no identifying or personal information)                      basic = basic hardware and network information                      full = all available information  -listDataTypes    lists all the available datatypes  -xml              generates xml output instead of plain text                    if redirected to a file with the extension ".spx"                    the file can be opened in System Profiler.app  -timeout          specifies the maximum time to spend gathering information                    the default is 180 seconds, 0 means no timeout  Redirect stderr to /dev/null to suppress progress and error messages.Examples:  system_profiler     Generates a text report with the standard detail level.  system_profiler -detailLevel mini     Generates a short report without identifying/personal information.  system_profiler -listDataTypes     Shows a list of the available data types.  system_profiler SPSoftwareDataType SPNetworkDataType     Generates a text report containing only software and network data.  system_profiler -xml >MySystem.spx     Creates a XML file which can be opened by System Profiler.app--------Luis Marques

exelban, 2021-11-24:
Ok, it's a reason why it cannot be shown in the app. `system_profiler` on 10.13 does not have `-json` option. Sorry, but it will not be fixed. It will require rewriting the parser from JSON to XML. And `10.13` is not supported.
I will take a look later today on the network.

exelban, 2021-11-25:
Yes, there is a bug with network details, will be fixed.

closed 2021-11-28 by exelban