exelban/stats #714: Stats 2.6.19 phones home to developer's personal server (leaking usage/launch activity) on startup without consent

sneak, 2021-11-23:
**Describe the bug**

The app contacts the developer's personal server on launch without consent.  This silently shares the user's IP address and launch history without the user's knowledge.

[image]

**Details:**
 
- Application version: 2.6.19

exelban, 2021-11-23:
Hi.
Yes, the app shares the anonymous stats on the server. Why? Because I want to know how my app is used. I don't have stats about downloads, so it's only one way how I get information about installs. You are free to block this event. It will not block the app's work, or you can build your own binaries with the disabled reports (just remove these [lines](https://github.com/exelban/stats/blob/4b1104dfc57e69b25a2977de76e692498c1db51b/Stats/AppDelegate.swift#L62-L65)). You can also run the Stats with `--omit` argument, it will omit sending data to the server.

closed 2021-11-23 by exelban

sneak, 2021-12-07:
> Because I want to know how my app is used. 

You don't think it's unethical to use your users' computers to spy on them this way without their knowledge or consent?

It's not *your* app.  It's free software, you gave it away.

You don't have any right to spy on the activity of your users, regardless of whether or not you think it's "anonymous".  (It's not anonymous - it transmits the user's client IP, and due to the lack of unencrypted SNI in a TLS negotiation, leaks to the user's ISP and upstreams that they are a user of your software.)

Why are you opposed to letting the user choose if they want to participate in your phone-home scheme (that violates their right to privacy) or not?  Give the user a modal on first launch asking for permission.  Simply phoning home without notice or consent is a violation of the user's privacy.

sneak, 2021-12-07:
Note that the developer/publisher of this software has rejected the removal of the spyware feature:

https://github.com/exelban/stats/pull/742

exelban, 2021-12-07:
Note that @sneak has aggressive posts. And don't propose anything. Just come and require to delete part of code. Without even asking why it is needed, and without trying to understand what data the app share with the server.

If someone has a question about what type of data Stats share with the server:
- app version
- app build
- which modules are enabled
- system version
- system language

Nothing else is collecting. And there is no personal data.

@sneak I propose you create a fork of Stats and do whatever you want. And yes, you are blocked.