exelban/stats #716: 10.13.6 checkbox issues

jonsummers, 2021-11-23:
**Describe the bug**
On 10.13.6 settings and app checkboxes are not shown. 
 
**Details:**
 - Device: Macbook Pro 13" 2015
 - macOS: 10.13.6
 - Application version: 2.6.20

The checkboxes are the setting panel at top right, all settings.
And the app panel checkboxes "Show icon in dock" and "Start at login"

Correct behaviour (checkboxes show) on 10.14.4 and (as sliders) on 11.6

**Fix:**
I believe the fix is as follows:
button.isTransparent = false // was true ; as in func toggleSettingRow#
button.wantsLayer = true // added (maybe not required) copied per func toggleSettingRow##

in Kit/module/settings.swift: line 121 in func header
in Stats/Views/AppSettings.swift: line 244 in func toggleView

Fix does not change 10.14.4 behaviour and cannot affect 10.15+
#, ## reference is Kit/extension.swift func toggleSettingRow #line 256 and ##line 258

jonsummers, 2021-11-23:
Further, I believe the NSButton ctr in in Stats/Views/AppSettings.swift: line 239 in func toggleView
            let button: NSButton = NSButton(frame: NSRect(x: 0, y: 0, width: 30, height: 20))
can be safely replaced with
            let button: NSButton = NSButton()

exelban, 2021-11-23:
Hi. Thanks for the feedback.
Does I understand correctly, the checkbox does not shown in the settings?

jonsummers, 2021-11-24:
The 'master' checkbox that controls the enabling of the panel at the top right of each settings panel does not show.

exelban, 2021-11-24:
Could you please share the screenshot of the bug?
I don't have 10.13 to check this.

jonsummers, 2021-11-26:
Here are Settings Panel and AppSettings images from v2.6.20 on macOS 10.13.6

[image]

[image]

jonsummers, 2021-11-26:
And here are the same panels on macOS 10.13.6 with fix "button.isTransparent = false" applied
(The "642" version number is my own for tracking my internal development changes)

[image]

[image]

exelban, 2021-11-26:
Thanks, will be fixed.

closed 2021-11-28 by exelban