exelban/stats #725: external SSD bug

P0pCat, 2021-11-27:
**Describe the bug**
SSD widget doesn't show the right capacity of external hard drive(it shows 100/100%), even tho I erased the drive.
*maybe the problem is in the drive, but I don't think that utility disk or time machine can lie

**Details:**
 - Device: MacBook Pro 2020 M12
 - macOS: 12.0.1
 - Application version: 2.6.20
[image]

exelban, 2021-11-28:
Hi. Thanks for the feedback.
What the file system of external hard drive?

P0pCat, 2021-11-28:
> Hi. Thanks for the feedback.
> What is the file system of the external hard drive?

APFS, also thx for such good app 👍

P0pCat, 2021-11-28:
oh wait it seems that it fixed itself somehow idk (I was trying to fix my "waiting to finish backup" bug and used first aid, didn't help but maybe caused to fix it)
@exelban

P0pCat, 2021-11-28:
Oh wait, excuse me I think I found an issue. Your app has some problem with detecting size of hard drives that are used as backup. Right after I started backup it showed that the capacity again is 100/100.

exelban, 2021-11-28:
If the disk is used as a time machine, it returns always 100% usage throw the API that I use.

closed 2021-11-28 by exelban