exelban/stats #726: Battery widget: link to battery settings/stats page

holdit, 2021-11-29:
I think it would be useful for the battery widget drop down to have a link to open macOS' battery settings window as it provides information about how the battery is being used. It also allows users to change power settings.

Maybe the battery icon could be clickable if you don't want to add more icons/text in that widget?

This was introduced with Big Sur, so it wouldn't work on Catalina or lower.

[image]

exelban, 2021-11-29:
Hi. Thanks for the proposal.
I will take a look at this feature. I suppose the Activity Monitor could be changed to this window in the battery module on the macOS 11+.

exelban, 2021-12-09:
Hi. I tried to implement this. I'm not sure that it's a good change. 
Battery settings are mostly for controlling. But in the Activity Monitor you could monitor the battery drain. And this app is more about monitoring rather than control.

closed 2021-12-21 by holdit