exelban/stats #731: Adding two elements to menubar increase spacing of all elements

Universumgames, 2021-12-01:
When I'm adding at least two element to the menubar from Stats, the spacing is completely off in Monetery...
No element:
[image]
Adding Ram usage and CPU:
[image]
>
When it's just one element, everything is just fine...
[image]

Universumgames, 2021-12-01:
Oh, maybe that's important: I'm using one of the new M1 Pros with the notch on the newest macOS Monterey version (currently enrolled in the beta program, but problem already exited before enrolling)

exelban, 2021-12-01:
Hi. Thanks for the reporting.
I'm not facing this bug on macOS 10.15, 11, and 12. Looks like it's something on the new MacBooks Pro. According to the screenshots, a new element increases the margins between all elements in the menu bar. It's for sure not a Stats bug.

thousandsofthem, 2021-12-01:
Seen this as well. MBP14, MacOS 12. Looks like system's totally-not-bug "feature"

exelban, 2021-12-02:
@thousandsofthem Hi. Do you have a new MBP14 (2021 with M1 PRO/MAX)?

thousandsofthem, 2021-12-02:
Yes, base new MBP 2021 14" with M1 Pro

thousandsofthem, 2021-12-02:
I've clicked around enabling and disabling various items and it's weird. system changes element width to compress (or move around in a more weird way) more stuff into right section of menu bar, after notch. After limit, it doesn't show any additional enabled items. On external display it does show all the items

"internal" screenshot, notch is where the gap between "Вкладка" and "Окно" is.

[image]

[image]

thousandsofthem, 2021-12-02:
The difference is ram round widget

[image]
[image]

exelban, 2021-12-02:
Looks like it's a system bug in the new MacBooks 2021.

closed 2021-12-02 by exelban

exelban, 2021-12-19:
Hi. There are screenshots in #757. According to them the macOS in some situations changing the margins of elements in the menu bar. Starting from macOS 11, each element in the menu bar has spacing. And it looks like macOS changing this space.