exelban/stats #732: [Suggestion] A few ideas to improve the look of the app

Squareheron942, 2021-12-01:
This app is pretty useful, there's only a few things I'd like to change:
1. Making the different menu bar icons more visually similar. What I mean by that is that, for example, the CPU usage icon has a small tag at the top saying what it is, and then the percentage used underneath, where you can colour it with many accents. But, the fan icon only shows the rpm it's spinning at, without the tags saying what it is, and also without the ability to recolour it the same way as you can with the CPU. The same goes with the sensor data. Although it's probably more difficult, I'd like to see each stat be its own icon to move around, as well as naming whatever the stat represents.
2. Making the icon size scale with the text size. What I mean is that right now, certain icons are up much more space than the text they contain does. It would probably save a lot of space on the menu bar to do this as well.

exelban, 2021-12-01:
Hi. Thanks for the feedback.
1. Colorizing the value of the sensors is not possible. Because for some sensors 30 is already red, and for some 90 is ok. Putting the sensor's name in the menu bar is not possible.
2. The menu bar items have the smallest size that allows fitting the content (text or chart).

closed 2021-12-01 by exelban

Squareheron942, 2021-12-01:
Thanks for clarifying that, I don't know much about Xcode so I didn't really know about these problems. I hadn't really thought about the other types of content (graphs as opposed to text) so it makes sense that the menu bar items are bigger.