exelban/stats #734: Allow RAM module to display memory pressure

aviexk, 2021-12-03:
Currently there is no way to display memory pressure, which is now the primary measure of memory performance for OSX.

from [apple docs](https://support.apple.com/en-gb/guide/activity-monitor/actmntr1004/mac):
> Memory pressure is determined by the amount of free memory, swap rate, wired memory and file cached memory.

from [stackoverflow](https://apple.stackexchange.com/questions/202744/what-is-memory-pressure-and-how-do-i-relieve-it):

> Memory Pressure is a better indicator of the health of the virtual memory system than the traditional UNIX measure of free, inactive and active pages.
> 
> For a brush up on the previous measure, have a read at:
> 
> [Isn't Inactive memory a waste of resources?](https://apple.stackexchange.com/questions/67031/isnt-inactive-memory-a-waste-of-resources)
> 
> Now, the caching of files is drawn as a good thing (green memory pressure) and you will only see the pressure raise in the main window of Activity Monitor when you have a large proportion of wired memory or the system starts swapping RAM to storage and evicting too many cache pages.

exelban, 2021-12-03:
Hi. Thanks for the feedback.
I agree that pressure level is better than utilization on macos. You can colorize the RAM value based on pressure. Also, you have a pressure level in the popup view.

closed 2021-12-03 by exelban

aviexk, 2021-12-03:
> Hi. Thanks for the feedback. I agree that pressure level is better than utilization on macos. You can colorize the RAM value based on pressure. Also, you have a pressure level in the popup view.

SUPER! didn't know we could colorize the RAM value! cheers