exelban/stats #735: [Feature Request] Show GPU processes

jcastro, 2021-12-03:
**Describe the bug**
Would be great to have a list of "Top GPU processes", just like we have on the CPU menu
[image]
[image]

exelban, 2021-12-03:
Hi. As I know there is no option to get this data from the system.

closed 2021-12-03 by exelban

jcastro, 2021-12-03:
[image]
Looks like it shows on Activity monitor, you can sort by GPU usage. Not sure if this is helpful